Ultimaker/Cura #11872: [5.0 beta] Invisible "dragable thingy" in the profiles dropdown menu with the Dark theme.

printingotb, 2022-04-22:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

Open the profiles drop down menu. The thingy that I unfortunately do not know the name of :sweat_smile: (please enlighten me) is completely invisible.
If you hover the mouse cursor over the scroll list, it becomes very faintly visible.

### Actual results

Invisible " thingy"

### Expected results

Visible "thingy"

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]

Vandresc, 2022-04-22:
Hi @printingotb, I've created a ticket for this, we will pick it up, hopefully, for the stable release. Do you mean the scrollbar :) ? 

Thank you for your feedback!

printingotb, 2022-04-22:
Great!
Thanks that's it! To be more precise, what you click and drag in the scrollbar is called thumb according to google.

fvrmr, 2022-04-26:
Devs see CURA-9186

DvoraNoob, 2022-05-01:
I prefer the dark color of Cura 4.13.1 than Cura 5.0, I think the color of 4.13 is less cloying, but it would also be nice to be able to choose the colors.