Ultimaker/Cura #11873: [5.0 beta] Dupicate themes

printingotb, 2022-04-22:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

Open dropdown to select Themes

### Actual results

There are two of each theme in the list.

### Expected results

Should be only one

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Vandresc, 2022-04-22:
Yes, we are currently working on this issue, should be fixed for the stable. Thank you for your feedback and for giving the beta a try.

printingotb, 2022-04-22:
Happy to help, keep up the good work!

fvrmr, 2022-04-26:
Devs see CURA-9147

nallath, 2022-06-15:
Fixed for stable (we just forgot to close this, sorry!)