Ultimaker/Cura #11874: Cannot open or save files with Beta 5 in Ubuntu

chipmunk1977, 2022-04-22:
### Application Version

5.0 Beta

### Platform

Linux - Ubuntu

### Printer

Custom Printer - (Creality CR-X)

### Reproduction steps

After running the Appimage file, I got a series of messages about plugins needing to be removed.  Followed those instructions.  Final message regarding configuration errors was as attached.  I could not open a file using the Open File menu item or the button, but drag and drop worked, as did Open Recent Files (which appears to be a list carried over from a previous installation).
It was possible to use and change slicing profiles, but it was not possible to use the Save to Disk function - nothing happens (no dialogue appears) but the system doesn't freeze.  Save Project is the same.  I have several instances of Cura installed (including Betas) and all work as expected.
Closing the configuration error dialogue and clicking on reset both appeared to do nothing, even with a restart of Cura after those actions.
I deleted the Appimage file and the folder named 5.0 in ~/.config/Cura
I then downloaded the file again and went through the same process with exactly the same results.

### Actual results

It is possible to continue working with the slicer, but impossible to save anything either as gcode or project.
The Configuration Error notice appears after every restart and closing it or using the reset button has not effect.  (For clarity, clicking on the Reset button generates the following warning - "Reset will remove all your current printers and profiles! Are you sure you want to reset?"  Clicking on yes or no has no effect, though I can continue working with the slicer.

### Expected results

That i can open files using the menu item, and save files.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

breeegz, 2022-04-22:
Ubuntu 20.04 Mate has same issue.  I can open the configuration folder under the Help Menu and it opens 2 folders, ~/.local/share/cura/5.0 and ~/.config/cura/5.0, but the "Save to Disk" button isn't saving.

Clicking on "Save Project" brings up the project summary, but not a save dialog.

jffmichi, 2022-04-22:
Same issue on fully updated Arch Linux.

Workaround:
Insert a USB-stick which gives you a working "Save to Removable Drive"-button.

chipmunk1977, 2022-04-22:
Good catcht with the USB Stick!  Can confirm that also functions as a workaround in Ubuntu.

gleonard3, 2022-04-22:
Same.  Chromebook Linux (Debian)

breeegz, 2022-04-22:
I am not able to experiment at the moment, but I wonder why..  mount path?
format?

jellespijker, 2022-04-26:
This sound familiar with the behaviour some of the developers where experiencing. Setting the `QT_QPA_PLATFORMTHEME` in the AppRun seemed to solve that, but it might have broken it for other Linux desktops.

It would be help to know the desktop environment you guys use.

@fvrmr can you add to the ticket a reminder to make the value of the following export as an argument with a default value? That way we can at least provide people with a hotfix if this was indeed the culprit.
https://github.com/Ultimaker/cura-build-environment/blob/a179158890d65583d97ab9d03454fd498c54ce75/packaging/AppRun#L10

We should also keep in mind that we might need to compile Qt6 and PyQt6 from source if this problem persist. Since the old file dialogs we used, made use of the ListDir QML object. Which is unfortunately not part of the precompiled PyQt6-Qt6 binaries

another alternative solution is to to use the same way how we construct the FileDialog (in Python) in the USB Writer

jffmichi, 2022-04-26:
> It would be help to know the desktop environment you guys use.
I'm using KDE with X11. Plese let me know if you need anything else.

breeegz, 2022-04-26:
Ubuntu 20.04 MATE

chipmunk1977, 2022-04-28:
Ubuntu 20.04 with XFCE desktop
also Ubuntu 22.04 with Plasma

wernight, 2022-05-01:
For Crostini using the AppImage I tried setting `QT_QPA_PLATFORMTHEME` to `qt5ct` and a few others without any effect.

I don't think it's related, but the initial welcome tutorial (shown after each upgrade) keeps showing again and again each time.

wernight, 2022-05-01:
Workaround for Crostini (and probably a bunch more):

1. Start Cura without the `export` from `AppRun`:
     1. `./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage --appimage-extract`
     2. `cd squashfs-root`
     3. `./Ultimaker-Cura`
2. Drag & drop the STL to Cura to load a model (the load/open dialog will still not work)
3. Save and export dialog will now work

jellespijker, 2022-05-01:
@wernight this is an indication that the `export` we set in `AppRun` is the root cause of this. Can you try setting them one by one, to narrow down the one causing problems?

breeegz, 2022-05-01:
I'm no expert, and this is the first time I have ever exported an appimage, but I'm willing to try, what's the worst that can happen.  

I tried to run Ultimaker directly like wernight, I can open files with drag-n-drop.  I can get my file browser to open with File-Open from the menu, but the my file browser hangs.  If I try to slice and save from the "Save to Disk" button, my file browser starts to open, but hangs/locks up like the File-Open from the menu.  So I'm not really functional with this workaround.  

If I make changes to ./AppRun, do I need to repackage Ultimaker in order for those to take effect?  What would be the preferred way to repackage the app in order to make changes?

I tried running AppRun with this extra line, 
echo $(dirname $0) and I'm getting "." so it looks like the export is trying to get to "./qt/plugins" (and all the other exports) but those folders doesn't exist for me.

Of course, I could be getting beat by my own newbness.

wernight, 2022-05-01:
@breeegz I'd just extra (using `--appimage-extract`) and change the `AppRun` file locally in the extracted directory for now.

@jellespijker Only removing `QT_QPA_PLATFORMTHEME` on `AppRun` or setting it to `gtk2` (default value in my Crostini environment) works.

jellespijker, 2022-05-01:
`QT_QPA_PLATFORMTHEME` was added because on other systems the file dialogs wouldn't show. It seems we need a more intelligent way of determining when to set it to which value.

wernight, 2022-05-01:
I agree. Still the 'load" dialog remains broken for some reason. Good that drag&drop works (frequently it doesn't on Crostini).

pconroy328, 2022-05-18:
Just downloaded and tried V5. I too cannot save gcode after slicing.  Went back to 4.13.x and can save.  Have not tried inserting a USB stick to see if that's a workaround.

$ uname -a
Linux 5.13.0-40-generic #45-Ubuntu SMP Tue Mar 29 14:48:14 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

$ cat /etc/os-release 
PRETTY_NAME="Ubuntu 21.10"
NAME="Ubuntu"
VERSION_ID="21.10"
VERSION="21.10 (Impish Indri)"
VERSION_CODENAME=impish
ID=ubuntu
ID_LIKE=debian

dgsardina, 2022-06-02:
Same issue on Arch with KDE and intel video. I manage to start cura with the following but I am unable to get the open dialogues to work:
`QT_QPA_PLATFORM=xcb LD_PRELOAD=/usr/lib64/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-linux.AppImage 
`
Note, I have also extracted the appimage and tried commenting out QT_QPA_PLATFORMTHEME and setting it to gtk2, still failing.

MariskaMaas, 2022-06-03:
Hey all,

Thanks for reporting. Opening files is essential to using Cura, so this is very unfortunate. 
We are aware of the issue and added an issue to our backlog. 
I will close this issue since it seems to be a duplicate.

You can follow the progress at https://github.com/Ultimaker/Cura/issues/12266

closed 2022-06-03 by MariskaMaas

wernight, 2023-01-23:
For some reason on clean Crostini (Ubuntu-based) the file open doesn't show, however I found a way.

 - First use the `*linux.AppImage` (not the `*linux-modern.AppImage`).
 - `sudo apt install flatpak && sudo apt remove flatpak && sudo apt autoremove`

Yes installing and removing flatpak should be no op but it does fit it. I don't yet understand why. Here is what it installs with flatpak:

```
The following additional packages will be installed:
  bubblewrap gnome-desktop3-data libappstream-glib8 libavahi-glib1 libgnome-desktop-3-19 libgpgme11 libmalcontent-0-0 libostree-1-1
  libpipewire-0.3-0 libpipewire-0.3-modules libspa-0.2-modules libxkbregistry0 p11-kit p11-kit-modules pipewire pipewire-bin xdg-dbus-proxy
  xdg-desktop-portal xdg-desktop-portal-gtk
Suggested packages:
  avahi-daemon malcontent-gui accountsservice evince
The following NEW packages will be installed:
  bubblewrap flatpak gnome-desktop3-data libappstream-glib8 libavahi-glib1 libgnome-desktop-3-19 libgpgme11 libmalcontent-0-0 libostree-1-1
  libpipewire-0.3-0 libpipewire-0.3-modules libspa-0.2-modules libxkbregistry0 p11-kit p11-kit-modules pipewire pipewire-bin xdg-dbus-proxy
  xdg-desktop-portal xdg-desktop-portal-gtk
```