Ultimaker/Cura #11875: plugin not supported Cura 5 beta+1

2G3D, 2022-04-22:
### Application Version

5 beta+1

### Platform

win 10 pro 64bit

### Printer

Geeetech A10

### Reproduction steps

immagine schermata di start

[image]

### Actual results

chiuso le finestre popup

### Expected results

che vengano caricati tutti ivecchi plugin delle vecchia release 4.13.1

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nulla

TwomD, 2022-04-23:
i have the same problem.
and when i open the marketplace the only plugin in it is "Barbarian Units 2.3.2" and nothing else.
it also shows "Failed to load Packages: could not reach marketplace. Retry?"

nallath, 2022-04-25:
I've fixed the failed to load packages already in https://github.com/Ultimaker/Cura/commit/86534e78a05ec5c61ec362d4a5b67083fc108b2d

As for the plugins missing, we've done a major API update as so many things changed. This means that plugin authors unfortunately have to update their plugins. As you might imagine we try to do this as little as possible, but this is sometimes needed to ensure that things are ready for the future.

For the moment you can get some beta versions of plugins here: https://community.ultimaker.com/topic/40313-fieldofview-plugins-for-cura-50/

As always, if you like the work that plugin authors (that aren't paid / hired by Ultimaker at all!) do, please consider donating them something.

closed 2022-04-25 by nallath