Ultimaker/Cura #11877: [5.0 beta] Seam placement still not as good as in older Cura

printingotb, 2022-04-22:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

I sliced the benchy with the same base profile and seam settings. I realize there are a lot of differences between the versions but no matter how much I try changing the numbers, the seam do not want to align very good. See especially the chimney and the back corner closest to the viewer.

I also know that the wall print direction is reversed with the arachne engine but mirroring the seam placement does not help either.

### Actual results

Uneven seam placement

### Expected results

As before or similar quality

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

printingotb, 2022-04-22:
Project files:
[seam 5 beta.3mf.zip](https://github.com/Ultimaker/Cura/files/8539535/seam.5.beta.3mf.zip)
[seam 4131.3mf.zip](https://github.com/Ultimaker/Cura/files/8539534/seam.4131.3mf.zip)

printingotb, 2022-04-24:
UPDATE:
It seams that all Seam corner Preferences except "Hide seam" aligns the seam basically perfect
[image]
.

fvrmr, 2022-04-26:
Hi @printingotb thank you for your report.
Will bring it up with the team. Keep you posted!

MariskaMaas, 2022-04-29:
Thanks for reporting! 
I've added this example to the already existing CURA-8600

sovtek, 2022-05-31:
Is there any update on when this will be fixed?
I love cura and I'm sure a lot of effort goes into and has gone into making the software as good as is it is, but I'm a little surprised 5.0 was released with this bug. For those of us who require seams to be not hidden and printed perfectly in a specific place on a model, this is a deal killer. I'm still stuck on version 4 until it's fixed. I love how the new version includes small details so of course I'm hoping this seam issue is fixed soon.

printingotb, 2022-06-02:
This issue together with #11881 are the top most on my wishlist to be fixed soon and are what's keeping me from fully transitioning from older Cura.

soxhi8, 2022-06-06:
Just in case you need another example, the file in https://github.com/Ultimaker/Cura/issues/12460 exhibits the same issue.

tlhintoq, 2022-06-27:
Yet another example.  Not hard to produce.  Take any cylinder, vase, cup, sphere.  Same values in 4.13.1 versus a version with Aracne engine like 5.  Reported this during Aracne beta testing so its not like its new or a surprise.
This makes version 5 a total non-starter for my shop.  Nothing new in 5 is worth this kind of horrible and unreliably place seam that can totally destroy the sellability of a part.  This cup is a simple example but it continues through anything from tool mods, jigs through to cosplay armor.  I can't print a helmet that is going to have that kind of noisy seam for a customer and expect them to find it acceptable.
[image]

tlhintoq, 2022-07-06:
> The seam placement seems to actually be offset from the asked for coordinates. Model is centered, model is a Ø10 circular tube. The seam should appear centered in the X direction at X0, Y-10. Absolute or relative positioning are the same. Adjusting machine size does not affect it.

Any chance you rotated the part on the print bed?  0 relative to the part follows the part, not the placement.  So if you rotated it 30° then X=0 would be a to 30° to the print bed.

printingotb, 2022-07-06:
It's not rotated and in my understanding it does not care what orientation the part has. It simply uses the parts center as X0,Y0 instead of the buildplate 0.

tlhintoq, 2022-07-06:
> It's not rotated and in my understanding it does not care what orientation the part has. It simply uses the parts center as X0,Y0 instead of the buildplate 0.

Not rotated.  Cool.

But that bit about it not caring about rotation is BS from my experience.  Any time I tell it something like Y=200, you can see that the seam moves with the rotation of the part.  "Relative" really is relative *TO THE MODEL* on a low level, and not relative to the part visually as you see it on the bed.

[image]

tlhintoq, 2022-07-06:
The issue you have in your example however is this:

X = 0
Y = -10

Well... Where is that in your model?  Because it looks like its in the middle of space.  There's no wall at 0,-10.
You said the DIAMETER is 10.
Well your placement is therefore 5 beyond that.  A diameter of 10 is a radius of 5.  If the part is centered then the Y direction goes from +5 to -5 for a total of 10.

So Cura did its best to give you as close as it could to what you asked.  It went negative on the Y by 10 then started searching out someplace to put a seam.  It had to put is _someplace_ since the point you gave it is impossible.

printingotb, 2022-07-06:
Sorry meant Ø20. 10mm radius. 
I have never experienced the seam rotating with the model. I just tried it.

virtualgraham, 2022-07-20:
This is still happening in 5.1 and with 0.8mm nozzle, the seam is so large and messy it absolutely ruins the print.

tlhintoq, 2022-07-20:
> This is still happening in 5.1 and with 0.8mm nozzle, the seam is so large and messy it absolutely ruins the print.

Remember you can have 4 & 5 side by side. I do this an use 4 for anything were this kind of seam placement matters.
[image]

tlhintoq, 2022-07-22:
[image]

Personal guess... Seam placement insists on disabling the option to even choose "none".  Its insists on trying to place at "back".
Which I suspect it looks for where the *back* is in real time while slicing and comes up with a different number on every layer, so its all over the place.

In 4.13 we could actually set "none".  In 5.0/5.1 we can't.
[image]

Cesinha1979, 2022-08-09:
[image]

tlhintoq, 2022-08-09:
Hey @Cesinha1979 
I'm getting some attention from them on this other ticket for the same problem.
https://github.com/Ultimaker/Cura/issues/12800

Can you please - zip your Cura project file and attach it in a post on that ticket?
That way they can open your project and see your same results on their computers and their machines.  It seems to be the only way they take it seriously.

tlhintoq, 2022-08-26:
So... Cura 5.1... Another example of not being able to get a straight seam on a circular object.
simple funnel.  No amount of playing with settings will get it to align.
![image](https://user-images.githubusercontent.com/11351506/187
[Funnel.zip](https://github.com/Ultimaker/Cura/files/9436844/Funnel.zip)
002532-d93be752-7e21-4d0d-8fb5-5b38800fd981.png)
I'm doing this on an Anycubic KobraMax
[Funnel.zip](https://github.com/Ultimaker/Cura/files/9436845/Funnel.zip)
I've uploaded the funnel STL as well as the Cura 3mf

tlhintoq, 2022-08-30:
Some other observations.
A part that is only 69mm size in the Y direction.  Placed center of the bed.
Seam location of x300, y-1000 gets a nearly straight seam.
Seam location of x300, y-500 gets a seam that is jagged.

This kind of behavior is very confusing to the user.  If my part is only 69 big, half of that would be positive and half negative when doing relative values for seam placement; so a range of +-34.5.  How does a part with a range of +-34.5 require a seam value of 1000 in order to align?
[image]

[image]

Side note of seam confusion... maybe it points to a problem in the math algorythm.  Wouldn't an X value of 0 [x]Relative be down the middle left/right?  0 relative is the center, isn't it?  So how is it the seam is drawn to the left and right corners?
[image]
I'm starting to think the seam doesn't really go where you tell it.  Instead it *acts like* it goes to the closest change in extrusion vector.  In this last example the seam isn't going down the middle where you set it but wants to go to the end of the straight wall where it meets with the curve.

sovtek, 2022-12-28:
It's almost 2023 and some of us are still forced to use Cura 4 because of this bug. Is there any timeline for when it will be fixed?

tlhintoq, 2022-12-28:
@sovtek 
I've been trying the new 5.3.0-alpha Christmas release that's like 2 days old.
I'm not seeing any improvement here.
[image]

If you didn't know, the resin slicer 'Lychee' now has FDM in beta.  Its not feature-rich yet, but for things I can use it on **I do**.  Nice to use one slicer for both my resin and FDM printers.

I'm not saying its a replacement for Cura yet.  Yet.  But if you're already being held back by the seam and can't upgrade anyway... Might be worth keeping an eye on, play with it a bit in advance of migrating.

MariMakes, 2022-12-29:
Hey @sovtek,

The only changes we introduced in 5.3.0-alpha Christmas release compared to UltiMaker Cura 5.2.1 are those which are needed for the new supports. So keep in mind, this is not a sneak peek for Cura 5.3 (there are some really cool new features coming up) but a spotlight release highlighting this new version of tree supports.

We did a big improvement to the zseams here: https://github.com/Ultimaker/CuraEngine/pull/1762 
This fixed introduced a lot of improvements and is scheduled for Cura 5.3 stable. 

I took some of the mentioned models through an internal build and the seams are looking like this:
[image]

[image]

tlhintoq, 2022-12-29:
Looking forward to 5.3
Those do look a lot less bad