Ultimaker/Cura #11878: [5.0 beta] Negative outer wall inset values is accepted but does nothing

printingotb, 2022-04-22:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

Apply a negative value for Outer wall inset and slice.

### Actual results

Value turns yellow but can be used and sliced with.
A negative value does nothing. And there are probably no reasons it should?

Positive values work as intended.

### Expected results

A settings value that does nothing at all should turn red and prevent slicing unless it affects something I am not aware of?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

[inset5 beta.3mf.zip](https://github.com/Ultimaker/Cura/files/8539667/inset5.beta.3mf.zip)

fvrmr, 2022-04-28:
Hi @printingotb thank you again for your bug reports.
Will bring this up with the team. Keep you posted!

MariskaMaas, 2022-04-29:
We discussed the issue with the team. 
It needs a closer look. For internal reference it is CURA-9231

Pougnagna, 2022-09-10:
> We discussed the issue with the team. It needs a closer look. For internal reference it is [CURA-9231](https://ultimaker.atlassian.net/browse/CURA-9231)
Dear @MariMakes ,
Please could you inform about CURA-9231 current status?
Thank you in advance!

MariMakes, 2022-09-13:
Hey @Pougnagna,

This bug is still present on our backlog with the intention to fix it whenever it reaches the top of our priorities list. 
Can you share how this bug is hindering you?

Pougnagna, 2022-09-14:
Global preview of Surface mode slicing:
[image]

Front preview of Surface mode slicing:
[image]

Global preview of Surface mode +Support blocker modifying overlap in Cutting mesh mode slicing:
[image]

Front preview of Surface mode +Support blocker modifying overlap in Cutting mesh mode slicing:
[image]

As you can see, there is an indent where the Support blocker trim the object.
I wish Inset could accept negative values to fix this behavior of Support blocker, unless Support bloker function has itself an issue.

I'm printing an RC airplane in LW-PLA, we need tp print in Surface mode but sometime we have to add someTop/Bottom layer depending of the part.

Thank you in advance for your reply.