Ultimaker/Cura #11880: [5.0 beta] Popup descriptions do not show the entire text and can't be scrolled

printingotb, 2022-04-22:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

Hover cursor over a setting.
A popup description appears with scrollbars.

### Actual results

There is no way to scroll it as it disappears as soon as I move the cursor from the setting.
If I use the mouse wheel or arrow buttons it scrolls the settings instead.

### Expected results

Show the entire description or make it possible to scroll it somehow.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Vandresc, 2022-04-22:
Yes, on our radar. Working on it for the stable. Keep bringing them in :)

printingotb, 2022-04-24:
> Keep bringing them in :

You asked for it 😁

fvrmr, 2022-04-25:
Closing this issue since it is a duplicate of https://github.com/Ultimaker/Cura/issues/11834

closed 2022-04-25 by fvrmr