Ultimaker/Cura #11881: [5.0 beta] Gaps between skin and walls

printingotb, 2022-04-22:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

Sliced the benchy, see attached project

### Actual results

There is a gap between the walls and the skin. A few dots of infill is placed where there is room.

### Expected results

No gap.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Layer 38:
[image]
[skin gap.3mf.zip](https://github.com/Ultimaker/Cura/files/8539840/skin.gap.3mf.zip)

nallath, 2022-04-22:
Damnit. I thought we fixed that :(

printingotb, 2022-04-22:
Perhaps one of the other bugs will offer some clues.

fvrmr, 2022-04-25:
Hi @printingotb thank you for your report.
Will bring it up with the team. Keep you posted.

Sonicboom247, 2022-05-03:
i wasted so much filament trying to figure this out... turns out Filter out gaps between walls has been removed. 4.13.1 on left, 5Beta on right....
[image]
.

printingotb, 2022-05-24:
Still present in the stable release. The culprit is skin removal width.
A more extreme example with Top skin removal width set to 4mm (this is the top most layer):
[image]

casperlamboo, 2023-01-09:
Hi @printingotb thanks for reporting the issue and apologies for waiting this long to get back to you.

I investigated the issue but I expect this is _expected behavior_. This probably is not what you wanted to hear but allow me to explain my reasoning.

The skin are is calculated by looking at a certain amount of layers above and see what areas on the current layer are no longer within the model. For the benchy model a few layers above layer 37 looks something like.
[image]
Note the railing at the back side of the benchy model. As there is actual parts of the model above the area you highlighted I do expect these area's not to be filled in with skin.

Please let me know if you agree with my assessment, and if not let me know why these area's should be filled with skin anyway.

printingotb, 2023-01-10:
Hello there @casperlamboo!
Thank you for your reply but I can't imagine this is expected behavior. It should not remove parts of the skin of a flat area with nothing above. That is how the settings guide describes it and also how it worked in older Cura.

Here's an example with some rectangles: (Please note that this is the topmost layer of flat surfaces)

At 7mm removal width it starts eating away at corners of the skin of all parts. This is in my opinion not intended. 

If it was intended to remove flat skin with nothing above, I would expect it to only remove any areas that are thinner than the removal width entirely and not remove parts of an area that is wider than the removal width. 

[image]
[image]
[image]
[image]
[image]
[Various shapes.3mf.zip](https://github.com/Ultimaker/Cura/files/10381536/Various.shapes.3mf.zip)

casperlamboo, 2023-01-10:
Ah yes, I agree that that is indeed not intended behavior. Thank you for the elaborate explanation and the included project files. I will update our backlog ticket.

However, I do think that this is a separate issue from the benchy model. Once this is fixed I predict the benchy example will have the same "gap" in skin as there is "inside geometry" above this area.

printingotb, 2023-01-10:
> Ah yes, I agree that that is indeed not intended behavior. Thank you for the elaborate explanation and the included project files. I will update our ticket.

Great, thanks!

>However, I do think that this is a separate issue from the benchy model. Once this is fixed I predict the benchy example will have the same "gap" in skin as there is "inside geometry" above this area.

Yeah that makes sense.

closed 2023-01-13 by jellespijker