Ultimaker/Cura #11882: [5.0 Beta] No "Tool Grips" on the model (rotate, translate, mirror, etc.)

GregValiant, 2022-04-22:
### Application Version

5.0.0

### Platform

Windows 10 Home (onboard video OpenGL 4.0 in compatibility mode)

### Printer

Creality E3Pro

### Reproduction steps

Start 5.0
Load any model

### Actual results

When the model is selected it cannot be rotated, mirrored, moved, etc.
Support blockers have the same problem - no grips.
(Attempting to use "Force Layer View Compatibility Mode" - The "Preferences" dialog does not have an "Enter" or "Cancel" button.  I had to use the CloseX button on the titlebar to dismiss the dialog.  I couldn't tell if my selections were accepted or not.)

### Expected results

I should be able to alter the orientation and location of the model.
The grips may actually be there, but are invisible.  Clicking around the model (in places where I recall the grips being located) occasionally gets a reaction/movement from the model.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8540409/cura.log)

I just can't get a grip on it...
[image]

nallath, 2022-04-22:
Huh, that is really strange. Especially since you can see the missing areas in the outline, which are caused by the tool handles.

I've tried reproducing this on Linux, but they do show for me (even when I'm in compatibility mode). @Vandresc can you check if you can reproduce this?

GregValiant, 2022-04-22:
_"you can see the missing areas in the outline"_
I missed that.  Then what we have here is a clear cut case of invisible grips.
It would appear I'm not the only one.  [UM Forum](https://community.ultimaker.com/topic/40288-ultimaker-cura-50-beta/#comment-306253)

Vandresc, 2022-04-22:
No, not seen by us either! This is scary! 
I will create a bug for the missing Defaults and Close buttons from the Preference page.

GregValiant, 2022-04-22:
@Vandresc I think the Machine Settings box has the same "lack of buttons" problem.
[image]

fieldOfView, 2022-04-22:
The ToolHandle shader is broken for "compatibility mode" renderers (which is why @nallath and @Vandresc cannot reproduce the issue, since they do have a decent GPU in contrast to @GregValiant ). 

Fixed by https://github.com/Ultimaker/Uranium/pull/811. @GregValiant it is a fairly easy fix that you could apply locally.

GregValiant, 2022-04-22:
@fieldOfView you are on a roll and I am on a rotate.  Thanks again.
(That in spite of the disparaging remark concerning my diesel powered video card.  Vacuum tubes for it are getting hard to find ya know).
I think I'll close this bug report then.
[image]

closed 2022-04-22 by GregValiant

jellespijker, 2022-04-22:
@Vandresc I couldn't find the bug on our backlog. But once you have created it you can drag it immediately to QA. I just reviewed and component tested @fieldOfView his PR, which is now merged. See that wasn't so scary

fieldOfView, 2022-04-23:
@Vandresc if you want to reproduce the original issue, set the "Force layer view compatibility mode" option in the General settings (and restart Cura).

Vandresc, 2022-04-25:
@fvrmr, FYI we have CURA-9190 and CURA-9188 related to this ticket.