Ultimaker/Cura #11883: [5.0 beta] Minimum infill areas counts as one even though they aren't touching.

printingotb, 2022-04-22:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

See attached project
Look at layer 45 and the following layers

### Actual results

It seems that the two small areas at the back of the benchy on layer 45 is a part of the larger area at the top.
(I thought it was a tiny gap between the inner walls but it does not help to increase line width).
The inner wall lines behave as if it were skin there.

Increasing the min infill area value from 10 to 500mm² fills the hut but still not those voids.
Increasing to 2000mm² fills all areas.

### Expected results

Those areas should be independent from the large area.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]
[min infill area.3mf.zip](https://github.com/Ultimaker/Cura/files/8540741/min.infill.area.3mf.zip)

printingotb, 2022-04-22:
At layer 46, the walls on the left side are not touching themselves but they are on the right side.
Related or just an edge case?
[image]

printingotb, 2022-07-02:
Just wondering if you have seen this as you usually respond quickly 😃

Joeydelarago, 2022-07-06:
Hi, thanks for reporting this. I would expect these small areas to be filled independently. I will discuss this with the team!

Ghostkeeper, 2022-07-27:
It seems that CuraEngine doesn't re-split up the infill areas if the area was one piece before walls generation.

We've had to make the hard decision here to prioritise this low. We don't think it occurs very often in practice, and if it occurs it's not the end of the world.

We do have a workaround: You can place an infill mesh around that corner. You can then simply add an extra infill wall there to close up the gap. Not ideal and it could be hard to place it precisely enough to not affect lower or higher layers, but it'll do for a detail like this until we see the opportunity to work in a fix.