Ultimaker/Cura #11884: Camera Button does not have a icon

RDols, 2022-04-22:
### Application Version

5.00 beta+1

### Platform

Windows 11

### Printer

Ultimaker 3

### Reproduction steps

Go to monitor tab. It looks like the camera button is not present, but actually it is, just no icon. Clicking on it also gives the expected camera window.

[image]

When mouse hoovering:
[image]

### Actual results

Button is invisible

### Expected results

I see the camera button

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Tiny visualisation bug: The camera button in the Monitor tab doesn't have an visible icon, so it appears the camera button isn't there.

nallath, 2022-04-22:
Thanks for taking the time to test the beta!

It's on our backlog. Developers see CURA-9117

fieldOfView, 2022-04-23:
Fixed in https://github.com/Ultimaker/Cura/pull/11893

RDols, 2022-04-23:
Although i can't test the fix and CI build failed, i will close this issue. Will reopen if the it's not fixed in the next beta.....

PR #11893 (2022-04-26): [5.0b1] Fix missing icons since move to PyQt6
This PR fixes some missing icons in the Cura 5.0 UI. In Qt6, the source of an image needs to be a resolved URL, whereas in Qt5 it could be a relative path string. This PR also fixes a "deprecated theme icon" warning.

After fixing this for the missing camera icon in the monitor page, I found the rest via a search in the source.

Fixes #11884, CURA-9117, CURA-9195
closed 2022-04-23 by RDols