Ultimaker/Cura #11885: 5.0.0 Beta support issues

fast2611, 2022-04-22:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

setting a custom layer height for supports
setting support wall count to 0 with zigzag pattern
atemping to save a project file using more than one screen

### Actual results

independent support layer height seems broken currently, or at least it shouldnt round 0.5 down the chosen value of to 0.16.

Setting the support wall line count to 0 breaks too and generates an extremly thin line around the infil
(in general alot of the support settings seems to be broken)

opening popup windows pops them up largely or completly out of screen bounds

### Expected results

4.x behavior

pop ups should stick to the current screen

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

project file cant be generated to to issue no 3

GregValiant, 2022-04-22:
My setup is the same as yours (Ender 3 Pro and Windows 10).
The first problem "custom layer height" can I assume you mean "Support Infill Layer Thickness"?  When set to 2X layer height it works.
The second one "support wall count to 0 with zigzag pattern" I can't duplicate.
I don't have a second monitor.  Saving a project works as it should with a single-monitor laptop.  There have been issues with Cura and dual monitors though.

fast2611, 2022-04-22:
appolgies, gonna have to strike issue 1, the beta currently doesnt run very well on my system and got a bad view at the layers due to issue2, gonna at least include a screenshot of that this time 
[image]

fvrmr, 2022-04-26:
Hi @fast2611 thank you for your report. Could you share a project file (file - save project) and share it here as a .zip?
We currently have some tickets on our backlog related to external monitors. Could you share a screenshot or screencast what is happening for you?
And for next please fill in one bug report for one issue, it is easier for us to track the issues then!

closed 2022-05-22 by fast2611