Ultimaker/Cura #11886: No GUI in Zorin OS Linux

jaymcd, 2022-04-22:
### Application Version

5.0 Beta 1

### Platform

Linux

### Printer

FLSUN QQS

### Reproduction steps

Never opens GUI

### Actual results

No GUI

### Expected results

Visible GUI

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No GUI available in Zorin OS Linux (Current Version)

fvrmr, 2022-04-26:
Hi @jaymcd thank you for your report.
Could you share your log file? You can find your log file here: `$HOME/.local/share/cura/<Cura version>/cura.log`

jaymcd, 2022-04-26:
here you go;

fieldOfView, 2022-04-26:
Unfortunately, github strips out attachments from email replies. Please use the github web interface to attach your logs (or post a link instead of an attachment).

jaymcd, 2022-04-26:
[cura.log](https://github.com/Ultimaker/Cura/files/8564177/cura.log)
Here is my log from Zorin OS

nallath, 2022-05-24:
Sorry! The bot was broken for a bit and it's having some hickups starting up again.

MariMakes, 2023-10-19:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue about your Linux GUI not working. 

Could you try if it's resolved in the 5.5 Beta release 🎉 , you can download the Cura version here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)

Thank you 🙏

jaymcd, 2023-10-19:
It has been working in Linu xnow for a bit - all good. Thanks 

Jay

closed 2023-10-20 by MariMakes