Ultimaker/Cura #11894: Multiply The First Time Doesn't Work

weidnerj1, 2022-04-22:
### Application Version

5.0 Beta

### Platform

Windows 11

### Printer

Creality CR6 SE

### Reproduction steps

Only happens the first time each time.  Start Cura, open .STL file and , click item and right mouse multiply  selected and it has number of copies - 1 - and click Ok, nothing happens.  Have to select 2 then 1 for it actually add one more.

### Actual results

Nothing

### Expected results

A duplicate of the item selected.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Only happens the first time, once it has been done once, it will work like normal.

fieldOfView, 2022-04-23:
Good catch! Fixed in https://github.com/Ultimaker/Cura/pull/11906

RickoT, 2022-05-10:
This also appears to affect copy/paste

fieldOfView, 2022-05-10:
@RickoT you are going to have to be more elaborate than that. Please tell us what problem you have with copy/paste, in what dialog (all?).