Ultimaker/Cura #11896: Cura 5.0.0 crashes when printing via USB

baldricc, 2022-04-22:
### Application Version

5.0.0 beta +1

### Platform

Windows 10

### Printer

Creality Ender 3 modded

### Reproduction steps

printing 20mm test cube via USB to printer. Print stops and hangs part way through printing brim and Cura 5 crashes and exits.

### Actual results

restart Cura, try again, same thing happens

### Expected results

Test cube should complete printing
[cura.log](https://github.com/Ultimaker/Cura/files/8544345/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/8544381/stderr.log)
g

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

screenshot of cura when crash occurs, shows print at -43000% completed
[image]

fieldOfView, 2022-04-23:
The USB Printing functionality is not used by Ultimaker printers, so it does not get (much) testing, so thanks for testing it and reporting it with logs. The logs show the problem occurring, but unfortunately they do not show exactly _where_ in the code the error occurs. 

The actual problem is that Cura tries to put a floating point number where an integer is expected. Previous versions of Python/PyQt would just silently convert from float to int, but they no longer do that implicitly. So we're going to have to find out where this happens, and do the conversion explicitly.

fvrmr, 2022-04-25:
Hi @baldricc thank you for your report.
We have created a ticket on our backlog for this. Devs see CURA-9196

Megiddo62, 2022-04-27:
I am using a Tevo Tarantula and just downloaded v5 and it worked fine for my first print and then on the second one crashed after about 5 min. The whole program closed itself so I could not get any screenshots. Also not sure where to find the log files.

NNGB-kor, 2022-04-29:
If possible, do not try to print via USB.
It is an act of giving up all the special features and computational power of well-crafted firmware.

fvrmr, 2022-05-02:
We possible have a fix for this issue, but it is for us hard to test. Since our printers doesn't have USB. So we have a nightly build to test if the issue doesn't happen anymore. [Link to Windows build](https://storage.cloud.google.com/cura-builds/5.0/20220502/Ultimaker-Cura-5.0.0-rc%2B20220502084956-win64.exe?_ga=2.201661009.-1198632912.1617002787)

Please let us know if this fixes the problem and if not also share your logs with us!

Megiddo62, 2022-05-02:
The Link to Windows Build above for the "USB Fix"  says that "***************@gmail.com does not have storage.objects.get access to the Google Cloud Storage object."

jmusser1, 2022-05-02:
> The Link to Windows Build above for the "USB Fix" says that "***************@gmail.com does not have storage.objects.get access to the Google Cloud Storage object." I think you have to enable access for people to download and try it out. I would love to and post my results.

Same issue, I can not print from USB.  Crashes after a few minutes.  I can be a 2nd tester.

FlyingMasterChef, 2022-05-03:
I also wish to test this feature as I have the issue as well.  And I absolutely despise printing via the SD card so I will only ever use the USB unless a very special case. I would never use the SD card for all my printing.. VERY inconvenient in my and many others situations.  I also get the does not have storage.objects.get access error when I click the link.  flyingmasterchef@gmail.com

fvrmr, 2022-05-03:
Sorry my bad, I updated the link above with the correct link.

DaveMeade5, 2022-05-03:
That patch worked for me on a small test print. It usually crashes very early in the print but it all completed succesfuly via USB
Printed on and Ender 3 S1
Well done guys. Thanks

jmusser1, 2022-05-03:
Confirmed, no crash 2hr print.

Megiddo62, 2022-05-04:
Yep just printed with the recent version of Cura 5 and it works just fine now via USB.

baldricc, 2022-05-04:
Can confirm printing from USB now working for me. Thank you for the fix.

cmiles68, 2022-05-05:
Confirmed usb printing fixed on Ender 5 - Thank you so much!

FlyingMasterChef, 2022-05-05:
It's working perfectly on my Anycubic Chiron now, via USB.  Did several 2+ hour prints and had no issues.

Greg

fvrmr, 2022-05-05:
Thank you all for testing! 
I will close this issue since it is fixed for the 5.0 stable release.

closed 2022-05-05 by fvrmr

Sonicboom247, 2022-05-06:
the nightly build is what i have been looking for! now i might be able to come back to Cura

Sonicboom247, 2022-05-11:
There should be a link to the nightly build repository, that way users can have issues fixed as soon as possible

frankygoop, 2022-05-15:
I was also able to print over USB with the nightly fix ONCE GET CONNECTED... and that is the problem. I have to try all USB plug back and forth hundred times until one finally connect to the printer. With Cura 4.XX always worked fine first time. I have a Tevo Tarantula, windows 11.
Where can we try more nightly?.
Thanks

fvrmr, 2022-05-17:
The stable 5.0.0 is released now: https://github.com/Ultimaker/Cura/releases/tag/5.0.0!

GDRPrint, 2022-05-20:
Confirm the above rc fix works, 
just noticed your above post to new version.
Many thanks for all your help !!
Thanks to everyone that posts.