Ultimaker/Cura #11897: Multiply Selected x1 does not work

pastera, 2022-04-22:
### Application Version

5.0.0-Beta+1

### Platform

Windows 10

### Printer

Tronxy X5SA Pro and Creality Ender 3

### Reproduction steps

Place single STL on buildplate
Right click and select "Multiply Selected" 
Leave quantity at 1 - Select Okay

### Actual results

Result: no new object

### Expected results

Object is copied

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

If you change the quantity to anything other than 1 it works correctly even if you return the value to 1

fieldOfView, 2022-04-23:
Good catch! Fixed in https://github.com/Ultimaker/Cura/pull/11906

fvrmr, 2022-04-25:
Closing this issue because it is a duplicate of https://github.com/Ultimaker/Cura/issues/11894

PR #11906 (2022-04-29): [5.0b1] Fix creating a single copy of a model
This PR fixes the Multiply selected models when the number of copies is left to 1. The `value` of the spinbox does not get initialised to 1, so even though the label says 1, the value is 0. The command was making 0 copies.

Fixes #11897
commit fcad2101 (2022-04-29): Fix creating a single copy of a model

Fixes #11897
closed 2022-04-25 by fvrmr