Ultimaker/Cura #11899: "Pick a screen color" missing

guate6, 2022-04-23:
### Application Version

5.0.0

### Platform

Windows 10/11

### Printer

Ultimaker S5

### Reproduction steps

Opened the material settings in Cura to add a filament and color. 

### Actual results

It has pre-set colors and hex only, no option to pick screen color

### Expected results

That the "pick screen color" option would be there, but it's missing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Please re-add this as it's a useful feature. Thanks.

fvrmr, 2022-04-25:
Hi @guate6 thank you for your report.
I know that this menu is updated because of the Qt6 upgrade, I will check with the team if this is expected. Keep you posted.

guate6, 2023-05-26:
Hello. Yes, this problem has not been fixed in any of the version 5
releases. It's a picture that was available in the older versions.

MariMakes, 2023-06-05:
Hey @guate6,

👋 I'm a fresh set of eyes reporting for duty 💪 
I don't make new materials very often 🤔 

Can you help me understand why this feature is important for you?
What are things you cannot do now that the feature is missing? 
What is something that the old feature used to do, that you cannot do with the hex code field?

guate6, 2023-06-05:
Precision of color for client previews is one thing (yes, my monitor is
color calibrated). Rather than having to manually input the hex code for
color.
When any software is released, once you get accustomed to a feature, when
it's missing from any newer versions, it's disappointing that the whole
thing seems to have gone a step backwards.
This became increasingly necessary, because the backup system is broken,
and sometimes I have to reset my profiles. So this taught me to have a
backup xls file with profile information, including colors. I was able to
go in and click the screen color on each cell to get each color done
correctly, but now that the "pick screen color" option is missing, I have
many extra steps to program a color for each material I add or restore to
Cura.

Raúl E. Fernández
Office (call/text): 352-573-6114
Mobile (call/text): 352-234-6161

guate6, 2023-06-05:
> Hey @guate6 <https://github.com/guate6>,
>
> I'll bring it up to the team to see what they can do to improve it.
> But I expect that if it's not a quick fix that it will not get a high
> priority to implement because last time I asked it was harder to bring it
> back than expected.
>
> Fingers crossed 🤞
>
> Also, can you tell me more about the backups not working on Windows?
> This is the first time I hear of this. Do you have a bug report for it?

I understand, and appreciate the positive attitude on this.
Personally I prefer "we will get to it eventually," rather than "we don't
care about this" that seemed to be the subtext of responses I received last
year. Thank you :)

Raúl E. Fernández
Office (call/text): 352-573-6114
Mobile (call/text): 352-234-6161

MariMakes, 2023-06-14:
Hey @guate6,

It seems like the feature you were looking for was not available in the library we are using at the time you reported this. But since then it has been updated and we can now implement https://doc.qt.io/qt-6.2/qcolordialog.html#details. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10687

Thanks for the report! 👍

guate6, 2023-06-15:
Great, thanks. I suspect in an upcoming update we might see this again?

Raúl E. Fernández
Office (call/text): 352-573-6114
Mobile (call/text): 352-234-6161

MariMakes, 2023-06-15:
Hey @guate6,

The upcoming release is super duper soon, so I expect it not to show up in that release.
But hopefully a later one, I cannot make any promises 😅

MariMakes, 2023-10-11:
👋 Quick update on our side. 

This is resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-11 by MariMakes