Ultimaker/Cura #11900: Can't login my Ultimaker Account

erickwill, 2022-04-23:
### Application Version

5.0.0

### Platform

Arch Linux Rolling

### Printer

Tronxy X5SA

### Reproduction steps

Opened Cura and tried to Sign in.

### Actual results

It didn't open the Google Chrome.
Here is a log:

Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Cyclic dependency detected between "file:///tmp/.mount_UltimaEoecEj/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaEoecEj/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
/opt/google/chrome-beta/google-chrome-beta: symbol lookup error: /usr/lib/libpango-1.0.so.0: undefined symbol: g_memdup2

### Expected results

To open Google Chrome to log in.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Cyclic dependency detected between "file:///tmp/.mount_UltimaEoecEj/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaEoecEj/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
/opt/google/chrome-beta/google-chrome-beta: symbol lookup error: /usr/lib/libpango-1.0.so.0: undefined symbol: g_memdup2

jellespijker, 2022-04-25:
Which desktop environment are you using?

jellespijker, 2022-04-25:
@tapir could you add your log files to it as well. Maybe even add the system logs when you press the sign-in button.
I would also like to know which desktop environment you're using

tapir, 2022-04-26:
@jellespijker I'm on arch linux latest Gnome and this is what it throws to console when I click the "Sign In" button

`/opt/google/chrome/google-chrome: symbol lookup error: /usr/lib/libpango-1.0.so.0: undefined symbol: g_memdup2`

tapir, 2022-04-26:
Arch doesn't have the Google Chrome in the official repos so I'm using the one from AUR which is just a repackaged version of the official Google debian packages 

```
source=("https://dl.google.com/linux/chrome/deb/pool/main/g/google-chrome-${_channel}/google-chrome-${_channel}_${pkgver}-1_amd64.deb"
	'eula_text.html'
	"google-chrome-$_channel.sh")
```

tapir, 2022-04-26:
Is it maybe an idea to use the standard library browser package? 

https://docs.python.org/3/library/webbrowser.html

jellespijker, 2022-04-26:
It looks like an issue with Chrome indeed. 

I myself use Chrome on a Manjaro, for me the external buttons also don't work, but that might be related to a different issue with the xcbcommon version which is packed in the AppImage.

Do you also have this issue with Cura 4.13.1?
If not a work around might be to log in there and copy the ultimaker authentication tokens out of the cfg file for the 4.13.1 config and paste them in the cfg for the 5.0.0

What happens if  you change the default browser?

tapir, 2022-04-26:
With 4.13.1 there wasn't any issues.
Changed the default browser to Firefox and still get an error

```
XPCOMGlueLoad error for file /usr/lib/firefox/libmozgtk.so:
/usr/lib/libcairo-gobject.so.2: undefined symbol: g_memdup2
Couldn't load XPCOM.

```

tapir, 2022-04-26:
I think the AppImage is using some archaic glib version

Nicknakin, 2022-04-28:
https://imgur.com/9pOqIGR
Running Plasma same issue

jellespijker, 2022-05-01:
It is compiled on Ubuntu20.04 which should have version 2.31, The upgrade was needed due to various reasons. See also #11863 
For compiling Qt6 we need at least glibc 2.28, While Ubuntu 18.04 carries 2.27, using other Linux distros for building Cura also gave various troubles, and using Ubuntu 20.04 was the best choice given the time constraints. 
We normally try to support an OS at a minimum until it's designated EoSS (End of Standard Support) by its maintainers. But we choose to not do this for Ubuntu 18.04 (EoSS will be April 2023) for the above-mentioned reasons.

tapir, 2022-05-01:
You could bundle the glibc with AppImage maybe?

jellespijker, 2022-05-01:
Bundling glibc is not considered good practice.

https://docs.appimage.org/reference/best-practices.html#binaries-compiled-on-old-enough-base-system

There are some methods we could use, but that will also mean that we have to add extra complexity and with it maintenance, to our build tools. Since we are using a minimum of C++17 and the required compilers coupled with Qt6 and Python3.10 in combination with older libraries. But I won't know how high fixing that will end up on our priority list. Especially if it is fixing it for a system which EoSS (End-of-Standard-Support) is approaching fast.

@fvrmr Maybe we should discuss this at the eCCB?

tapir, 2022-05-01:
Ultimately 5.0 will end up on official repos/packages for a lot of distributions so AppImage is not priority for linux users as well.
For me it's just a way to test the betas. Flatpak would solve this issue since it has proper dependencies and it's not just a bundled up ZIP but I could understand you don't want to put time on it.

jellespijker, 2022-05-01:
@tapir All of our developers use/prefer Linux so believe me when I say we want nothing more then to offer support for all those lovely distro's. If time allows it we will certainly do that.

I know that @thopiekar is working on a Flatpak for Cura 5.0, I don't know if he already managed to get it up and running, Since our recent changes also broke somethings for him.

phil2sat, 2022-05-02:
I personally love linux but since a couple of years, for me it gets very hard to live with. some programs needs very old deps, some very new, to get everything running most of the time i end up using unstable branch and compile things on my own. best examply is my ryzen laptop, has bugs with kernel 5.9 but amd driver module only builds on 5.9, took 2 days to get vulkan opencl and everything running on a 5.17 kernel. another 2 days for fusion360 and a day for cura 5.0. 10 years ago things went much faster and easier, the linux world needs to get defragmented. i dont like snap, flatpack, docker or appimages, they install rendundancies, and if it goes on we end up in installing a whole linux system per app. its like uh i need round corners, hey but the performance goes down, who cares buy a rtx3080, to get the filemanager running.

erickwill, 2022-05-17:
> @erickwill Which desktop environment are you using? and can you add your log files.

I'm using Gnome, under Arch Linux rolling.

`PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
libGL error: MESA-LOADER: failed to open crocus: /tmp/.mount_UltimaxYVPiu/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/libLLVM-13.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_UltimaxYVPiu/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/libLLVM-13.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
[erick@mobile-hp Apps]$ exec optirun /home/erick/Apps/Ultimaker-Cura-5.0.0-linux.AppImage
/usr/bin/vglrun: line 193: hostname: command not found
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Warning: Ignoring WAYLAND_DISPLAY on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
libGL error: MESA-LOADER: failed to open crocus: /tmp/.mount_UltimabHFD5Z/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/libLLVM-13.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_UltimabHFD5Z/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib/libLLVM-13.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
Cyclic dependency detected between "file:///tmp/.mount_UltimabHFD5Z/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimabHFD5Z/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
qt.qml.context: file:///tmp/.mount_UltimabHFD5Z/share/cura/resources/qml/Settings/SettingView.qml:427:16 Parameter "index" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
/opt/google/chrome-beta/google-chrome-beta: symbol lookup error: /usr/lib/libpango-1.0.so.0: undefined symbol: g_memdup2
`

Seems it breaks here: 
`/opt/google/chrome-beta/google-chrome-beta:` symbol lookup error: /usr/lib/libpango-1.0.so.0: undefined symbol: g_memdup2`

erickwill, 2022-05-17:
> Same on Kali linux, its the glib2 version which makes problems libglib2.0-0_2.72.1-1 ill guess libglib2.0-0_2.58.3-2 works
> 
> `exo-open: symbol lookup error: /lib/x86_64-linux-gnu/libpango-1.0.so.0: undefined symbol: g_memdup2 `
> 
> EDIT: Complied 5.0 myself and login works dont know if it works on other distros but on my kali i works, maybe glibc2.34 from unstable is nessesary as its a build dependency from cura, mhmm should have check the login on default beta after installing glibc2.34
> 
> Kali ships glibc2.33
> 
> Here is the Appimage: [https://drive.google.com/file/d/1LEmhirxYvEgrmdAuUB-Ywvu5yq9DoCmr/view?usp=sharing](url)

Looks good! Did you compiled the last stable one? Could you please share? Thanks

closed 2022-05-23 by github-actions[bot]

nallath, 2022-05-23:
Sorry for that. THe bot broke and now it's running again

PR #12736 (2022-07-12): Cura 9205 appimage not launching on ubuntu 2204
Building two AppImages, one based on Ubuntu 20.04 and one with on Ubuntu 22.04. This also solves the sign in button not working on some Linux systems. Keep in mind that this is more of an intermittent solution and that we should still try to solves this in a more generic/future proof-way

Fixes both CURA-9342 and CURA-9205

The results of this run can be found here: https://github.com/Ultimaker/Cura/actions/runs/2648505664

- Fixes #11856
- Fixes #12453
- Fixes #11900
- Fixes #12232
- Fixes #12291