Ultimaker/Cura #11903: Support error: unsupported area

alexshorokhov, 2022-04-23:
### Application Version

5.0 beta 

### Platform

windows 10

### Printer

Ultimaker original

### Reproduction steps

1. Uploaded cura v4.13.1 .3mf file into cura v5.
2. Sliced

### Actual results

Unsupported corner (image "Cura 5")

### Expected results

The same supports as in Cura 4 (image "Cura 4").

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]

Supports was ok when I sliced the same model from .stl file, not .3mf

[Cura 5 bug project.zip](https://github.com/Ultimaker/Cura/files/8546278/Cura.5.bug.project.zip)

fvrmr, 2022-05-04:
Hi @alexshorokhov thank you for your report.
I can reproduce this issue, will bring it up with the team. Keep you posted.

fvrmr, 2022-05-05:
I created a ticket for this. Devs see CURA-9250