Ultimaker/Cura #11907: [Cura 5.0.0 Beta +1] Cura does not start under Win10

SIeeepy, 2022-04-23:
### Application Version

5.0.0 Beta +1

### Platform

Windows 10, GeForce RTX 3090

### Printer

None (program doesn't start)

### Reproduction steps

Install Cura, open Cura.

### Actual results

The installation process seems to be successful. After starting Cura the loading screen appears ("loading machines", "loading UI"...), but after Cura switches to the actual larger working window, it stays completely white and does no longer work/respond. 

### Expected results

Starting Cura

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8546659/cura.log)

What I've tried to do so far (to no avail): Reinstall to different locations, uninstalling all previous versions,  disabling antivirus (Windows defender), deleting all configuration folders under %APPDATA%\cura\, update graphic card driver, update vc_redist x64, installing and opening as administrator.

Previous versions of Cura did work just fine.

chunter1, 2022-04-24:
Had the same issue and in my case the "Ultimaker-Cura.exe" file was missing after installation.
I resolved the problem by choosing a different installation directory (e.g. ...\Ultimaker Cura 5.0.0_test) during the installation process.
Hope it helps ;)

SIeeepy, 2022-04-24:
Thanks for the help chunter. I've checked, the "Ultimaker-Cura.exe" is not missing and I also have tried different installation locations. To no avail. :(
As mentioned the .exe will be executed and the new bright blue loading screen appears and lists all the things it's loading, but the final UI window freezes ("Ultimaker Cura is not responding").   

I think your problem was an aggressiv antivirus reaction. I'm using just the internal Windows antivirus. I've also tried to open Cura with deactivated "real time protection" of this Windows antivirus app, without success.

FransBothma, 2022-04-24:
Ihave the same Problem on Windows 11. also tried all of the points mentioned above to no avail.

ISHaDoW6I, 2022-04-25:
Me as well. Using Windows 11 Home and also have the above issues. Installs just fine but after loading everything it just hangs. Need help, was really looking forward to trying it.

DaveCarrera4, 2022-04-28:
Same issue with Windows 11 - new Alienware core i9 32 Gb laptop.  Splash screen finishes, main Cura window opens a couple seconds and the message; (Not Responding).  Then fan goes all the way to 100% and nothing happens, blank window.

SIeeepy, 2022-04-30:
I found out the starting problem for my case (but still have no satisfying solution). Cura 5 seems to have a problem with a multi monitor setup. If I disable output to my second monitor, I can start Cura 5 and run it stable.
I did not observe this monitor problem with any other Cura Version. The work around is also not a practical solution, as I’m not willing to sacrifice my second screen just to get an app started. 
There are some tips, e.g. to uncheck the box in the settings “Restore window position on start”. But this doesn’t help. Checked or unchecked, Cura 5 doesn’t start with a second monitor, previous versions showed no issue.

TheTabe, 2022-05-03:
> I found out the starting problem for my case (but still have no satisfying solution). Cura 5 seems to have a problem with a multi monitor setup. If I disable output to my second monitor, I can start [...]

Same issue here on Windows 11. Just a large white window when I launch it with an external monitor connected to my laptop but when I disconnect/disable the 2nd monitor it launches into the UI as expected.

FransBothma, 2022-05-03:
> I found out the starting problem for my case (but still have no satisfying solution). Cura 5 seems to have a problem with a multi monitor setup. If I disable output to my second monitor, I can start [...]

Thanks !! Same thing on my windows 11. At least now I can use it an start playing.

DaveCarrera4, 2022-05-04:
> I found out the starting problem for my case (but still have no satisfying solution). Cura 5 seems to have a problem with a multi monitor setup. If I disable output to my second monitor, I can start [...]

Same here.  Disconnected 2nd monitor and it loaded on Win 11 Alienware laptop.

whazzat4me, 2022-05-05:
Maybe related. My Cura 5 opens then freezes on the working page. I discovered if I press the Maximize button on the top right of the page it starts working.

nallath, 2022-05-05:
> Can we get this dual monitor glitch fixed? I am not willing to loose my second screen.

We are going to do our best, but we've had little luck getting it to reproduce.

evtrados, 2022-05-05:
In case you are using a docking station which connects to your laptop via a USB cable the issue could be because the GPU used on your native screen is NVIDIA and on the monitors connected to the docking station it is defaulting to the default GPU. At least that was the issue I was facing. If that's your setup too, I can add some steps that fix this (hopefully - because I was getting a crash instead of a non-responsive window).

rburema, 2022-05-05:
To everyone: Can you _also_ please include your log files? It'll maybe help us pinpoint the problem (or ascertain we're dealing with multiple superficially similar but unrelated problems).

For example, in the original logs attached, these could be 'red herrings' or vital clues:
 - I've noticed that the block with OpenGL vendor / GLSL version doesn't get printed, even though the log says it otherwise detected 4.1 (of OpenGL) -- that is very strange
 - Also in the original log is a(/that) line about `Could not restore to previous location on screen, since the sizes or number of monitors have changed since then`

TheTabe, 2022-05-05:
Here's my log
[cura.log](https://github.com/Ultimaker/Cura/files/8631298/cura.log)

SIeeepy, 2022-05-05:
> To everyone: Can you _also_ please include your log files? It'll maybe help us pinpoint the problem (or ascertain we're dealing with multiple superficially similar but unrelated problems).
>
> For [...]

Thanks rburema!
Here's a current log: [sleeepy_cura.log](https://github.com/Ultimaker/Cura/files/8631260/sleeepy_cura.log)
1) I'm not sure I understand.  AFAIK OpenGL is part of the graphic card driver. For testing I switched back to an older Nvidia driver (it didn't help) 
2) As mentioned I've experimented also with single monitor setup. So the "sizes or numbers of montitors have changed" seems to be a correct statement.

SIeeepy, 2022-05-05:
> In case you are using a docking station which connects to your laptop via a USB cable the issue could be because the GPU used on your native screen is NVIDIA and on the monitors connected to the [...]

Thanks for the tip. In my case it's just two identical monitors plugged into a standard Geforce GTX 3090.  Old Cura thinks this is as very fine setup, new Cura thinks it can't work under these circumstances.

rburema, 2022-05-05:
@DaveCarrera4  Thanks for the info, could you also please attach the log file.

ISHaDoW6I, 2022-05-05:
My case as well, using 2 identical 35” Samsung smart monitors

FransBothma, 2022-05-05:
> To everyone: Can you _also_ please include your log files? It'll maybe help us pinpoint the problem (or ascertain we're dealing with multiple superficially similar but unrelated problems).

I would like to help, where do I find the Log File ?\

SIeeepy, 2022-05-05:
> > To everyone: Can you _also_ please include your log files? It'll maybe help us pinpoint the problem (or ascertain we're dealing with multiple superficially similar but unrelated problems).
> 
> I would like to help, where do I find the Log File ?\

for me the path in Win10 is: C:\Users\[Username]\AppData\Roaming\cura\5.0  and the file name is cura.log

FransBothma, 2022-05-05:
> for me the path in Win10 is: C:\Users[Username]\AppData\Roaming\cura\5.0 and the file name is cura.log

Thank you. here is mine. hope it help
[cura.log](https://github.com/Ultimaker/Cura/files/8633402/cura.log)
[cura.log](https://github.com/Ultimaker/Cura/files/8633404/cura.log)
s

ISHaDoW6I, 2022-05-06:
Here is mine also. Sent from Mail for Windows From: FransBothmaSent: May 5, 2022 1:38 PMTo: Ultimaker/CuraCc: ISHaDoW6I; CommentSubject: Re: [Ultimaker/Cura] [Cura 5.0.0 Beta +1] Cura does not start under Win10 (Issue #11907) To everyone: Can you also please include your log files? It'll maybe help us pinpoint the problem (or ascertain we're dealing with multiple superficially similar but unrelated problems).I would like to help, where do I find the Log File ?\for me the path in Win10 is: C:\Users[Username]\AppData\Roaming\cura\5.0 and the file name is cura.logThank you. here is mine. hope it helpcura.logcura.logs

ScruffR, 2022-06-03:
The issue still persists with the official release and I am ***not*** have  a second monitor (MS Surface Pro 6 and Thinkpad Yoga 17 on Win 10 Pro 64bit)
[cura.log](https://github.com/Ultimaker/Cura/files/8833036/cura.log)

nallath, 2022-06-03:
> The issue still persists with the official release and I am _**not**_ using a secondary monitor (MS Surface Pro 6 and Thinkpad Yoga 17 on Win 10 Pro 64bit) [cura.log](https://github.com/Ultimaker/Cura/files/8833036/cura.log)

You have an entirely different issue. 

```
2022-06-03 16:23:39,619 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [606]: file:///C:/Program Files/Cura/share/cura/plugins/ModelChecker/ModelChecker.qml:8:1: Type UM.SimpleButton unavailable
2022-06-03 16:23:39,619 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [606]: file:///C:/Program Files/Cura/qml/UM/SimpleButton.qml:28:5: Type RecolorImage unavailable
2022-06-03 16:23:39,619 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [606]: file:///C:/Program Files/Cura/qml/UM/RecolorImage.qml:5:1: The plugin 'C:/Program Files/Cura/qml/QtGraphicalEffects/qtgraphicaleffectsplugin.dll' uses incompatible Qt library. (5.15.0) [release]
2022-06-03 16:23:39,619 - DEBUG - [MainThread] ModelChecker.ModelChecker._createView [127]: Model checker view created.
2022-06-03 16:23:39,619 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._createDialog [76]: Creating dialog [MoreInfoWindow.qml]
2022-06-03 16:23:39,635 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [606]: file:///C:/Program Files/Cura/share/cura/plugins/SliceInfoPlugin/MoreInfoWindow.qml:64:13: UM.Label is not a type
```

This pretty much looks like you install went wrong somehow. It's almost as if you're mixing two installs of Cura. 

Did you perhaps install Cura in the exact same location as where you had Cura 4.13 installed?

ScruffR, 2022-06-03:
@nallath, thanks for the reply and you are correct - I did :blush:

I never had an issue with doing that with any of the previous versions - additionally all the previous installs also noticed the fact and asked whether I want the previous version uninstalled before installing the new one, but the 5.0.0 installer did not.

jldavis1969, 2022-06-24:
I have this same issue except I have 6 monitors, 3 on a RTX 2060 Super, 3 on a GTX 960.

I have found on another forum (Cura github here) that if you unplug one monitor so that Cura 5.0.0 is moved to another monitor it will work.

In my case, Cura 5.0.0 was on my righthand lower monitor and would go not responding there.

I disconnected said monitor and got Cura 5.0.0 to fully load and work on my middle monitor.

My righthand monitor is connected via the DVI port (Maybe and issue with DVI), 
My middle and lefthand monitors are connected via the DisplayPort connector. 
All three monitors are connected to my RTX 2060 Super video card.

I would try my fix if you are using multiple monitors and getting the Not Responding issued described in the above message.

I have added this comment to the Ultimaker forum (Cura 5 Multi Monitor Crash)

Majormer, 2022-06-26:
For what it is worth, if I change my primary monitor to the monitor Cura has been failing to load on, it will launch fine (Windows 11).  Geforce RTX 3070 video card.  I run with a 4 monitor setup.  Otherwise, I have the same symptoms with a white, non-responsive window as described above.  Since having my secondary monitor running as primary is a non-sustainable solution except for maybe quickly klicking off a print job to my OctoPrint server, I'm hoping there is a solution in the future.

Joeydelarago, 2022-06-30:
One of our team members have managed to reproduce this bug. We will discuss it. 

Specifically the bug caused by trying to display Cura when using the GPU (Nvidia cards it seems). If you can plug you monitor into your motherboards HDMI slot (Not using GPU) and it works normally then this is the same issue you are having. 

Thanks for attaching all the logs, they help a lot!

pedrolamas, 2022-06-30:
Thanks for following up on this, @Joeydelarago!

Your findings do match my own setup as I use a Dell XPS 15 laptop (with an Intel integrated GPU and a NVIDIA GeForce RTX 3050 discrete GPU).

This has made me wonder if it would make any difference to force Cura to run on either of these GPUs instead of letting Windows automatically handle it...

In my own system, it seems I have it set to use the Nvidia GPU:

[image]

pedrolamas, 2022-07-03:
FWIW, changing from "high performance" (Nvidia GPU) to "power saving" (integrated graphics) has solved the problem for me: Cura 5 has not crashed a single time (nor cause the graphics driver to crash as it did) for the past few days!

rburema, 2022-07-21:
We're still investigating this. (So it we couldn't fix this in 5.1 yet unfortunately.) At present moment, we have two (other) leads, bot largely outside of our sphere of direct influence:

- A surprising connection was found to some particular audio drivers: https://github.com/Ultimaker/Cura/issues/12197#issuecomment-1157988232
- Found (and reported to) a possible bug in Qt, that might affect certain multi-monitor setups: https://bugreports.qt.io/browse/QTBUG-105070

wiwa5369, 2022-08-16:
I've also have this problem with dual monitors in V 5.0 and 5.1. 
My **solution** for this moment was to edit the cura.cfg found in Windows %APPDATA%\Roaming\cura\5.0 or ...\5.1
Using any Textedit-tool search the line content "windows_left =" and set the value to a x-coord near the outer left side of total screen coordinates, may be 10 or so. 
This works and cura starts on the left (1st) Screen/monior well.
Regards

wiwa5369, 2022-08-16:
Programmers may read the cura.cfg anyway on start. Why not put a quick for-the-moment-workaround in the code that forces the start on windows_left = 10 i.e. on the 1st monitor, later moves to any screen will work fine.
regards

wiwa5369, 2022-08-16:
Must revise my 1st comment a little. Cura freezes if I move it to the right monitor (whatever number it was) and do some essential changes i.e. change the print-density (high to normal or draft). Leave it on the left screen it works. As soon as it touches the second screen it freezes.
regards

wiwa5369, 2022-08-16:
Try this (my workaround):

**>**$line = Get-Content "C:\Users\**xxxxx**\AppData\Roaming\cura\5.1\cura.cfg" | Select-String "window_left = " | Select-Object -ExpandProperty Line
**>**$content = Get-Content "C:\Users\**xxxxx**\AppData\Roaming\cura\5.1\cura.cfg"
**>**$content | ForEach-Object {$_ -replace $line,"window_left = 1"} | Set-Content "C:\Users\**xxxxx**\AppData\Roaming\cura\5.1\cura.cfg"
**>**[System.Diagnostics.Process]::Start("D:\Ultimaker Cura 5.1.0\Ultimaker-Cura.exe")

1. Change **xxxxx** with your Userpath-name
2. Check and change all path- and filenames for your environment
3. Remove lineprompt "**>**" handle following text in one line
4. Save text as Powershell ".ps1" file
5. Exec with right-click -> open with powershell

Cura 5.xxx may start allway on coordinate "windows_left = 1"

regards

MostHostLA, 2022-08-17:
Was having very similar issues.
Deleting the roaming/cura folder got cura to open again same as disconnecting input from other monitors.

Safe to say the issue is very much still an issue.

wiwa5369, 2022-08-17:
Same to me. So I start Cura with my workaround now very fine until solution was implemented.

elradlako, 2022-08-20:
Hello. Did any of you find solution? I've used cura 5.1 for a week and then i didn't use it for 2 days. When i tried to launch it got blue loading window then opened to white screen and closed after 3 seconds. Was looking for help online tried to rebooting PC reinstalling cura reinstalling it to other location then tried to install previous version of cura then tried to install prusa slicer, slic3r etc. No slicer works anymore on my PC. And i just have only one display. I really hope someone can help me with this one

wiwa5369, 2022-08-20:
Sorry, no idea. Try again deinstall, delete all from %appdata%\Roaming\Cura\5.xx and try an previous Cura version (4.3-4.13 maybe works better?!).

elradlako, 2022-08-20:
> Sorry, no idea. Try again deinstall, delete all from %appdata%\Roaming\Cura\5.xx and try an previous Cura version (4.3-4.13 maybe works better?!).

Already done that. Uninstalled removed %appdata then installed again not even a bit better. Uninstalled deleted %appdata installed 4.13 didnt work either. Uninstalled installed prusa slicer didnt work etc.

wiwa5369, 2022-08-21:
Did you have a System Recovery Point from times before you get this problems then reactivate it. If this not helps or it's not true my ideas are at the end except a full restore or new Winsetup. Maybe you find in other forums some help - Good luck...

elradlako, 2022-08-25:
[cura.log](https://github.com/Ultimaker/Cura/files/9428604/cura.log)

here im attaching log so maybe anyone got an idea

wiwa5369, 2022-08-26:
I'm not a Cura programmer so I can't help with that. 
(Seems to be a corrupt install, cause folders and files are missed and printer was not found. Please wait for a spezialist answeres here or post into another or new thread. )

elradlako, 2022-08-27:
Okay so i just forced my pc to look for windows update and it found something updated and cura magically started to work after reinstall

elradlako, 2022-08-28:
okay so it worked for one day and today got same scenario. However i got no more updates from windows, already reinstalled it few times and cleared all cache. any ideas?

peterashuk, 2023-02-19:
Does anyone have a solution to running Cura 5.2.2 with dual monitors ?

TheTabe, 2023-02-20:
Stopping the Nahimic service (and setting  it to not auto start) worked for me.

MariMakes, 2023-03-31:
Quick update on our side for those who are still stuck. 

We might have a fix for some of the issues reported here. 🎉
It's a bit hard for us to test, but we have a nightly build available.
Can you please try it out and let us know if it resolves your issue?

[You can find the download links and the discussions here.](https://github.com/Ultimaker/Cura/discussions/15098)