Ultimaker/Cura #11909: Pop up tips box size problem.

Hammster56, 2022-04-23:
### Application Version

5.0.0 Beta+1

### Platform

Windows 10

### Printer

Anycubic Mega X

### Reproduction steps

Hovered over a setting in Cura.
Saw the tip box pop up.
Not everything could be read in the tip box. Additional info scrolled out of the box and I couldn't scroll the contents up or down in the box. 
It seems the box should resize to show all contents in the box.

### Actual results

[image]

### Expected results

I should be able to read the entire pop up box contents. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None.

Sonicboom247, 2022-04-23:
agreed, happening on mine as well on Win10 on tips with more than 6 lines of text, when fixed can you maybe move the box out of the way and to the right or top?

MinProwler, 2022-04-24:
I see this behavior as well with beta+1 in  MacOS Monterey (12.3.1, Intel cpu).

Cheers

fvrmr, 2022-04-25:
Hi @Hammster56 thank you for your report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/11834
I will close this issue since it is a duplicate.

closed 2022-04-25 by fvrmr