Ultimaker/Cura #11910: Cannot access marketplace in Ultimaker Cura 5.0.0

cmues, 2022-04-23:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

I opened the application
Opened marketplace
It says it can't reach the marketplace

### Actual results

Cannot access the marketplace plugins

### Expected results

Can access marketplace plugins

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Sonicboom247, 2022-04-23:
i would give it a week, i doubt many packages are compile for 5.0 yet

chipmunk1977, 2022-04-23:
This is not a bug.  The screenshot you included shows that you can indeed access the marketplace.
The plug-ins need to be updated to work with this beta release and that takes a while - they will appear there over the coming days and weeks.

nallath, 2022-04-25:
@chipmunk1977 is correct. You can already get some beta versions of the plugins here: https://community.ultimaker.com/topic/40313-fieldofview-plugins-for-cura-50/

closed 2022-04-25 by nallath