Ultimaker/Cura #11911: Raft not calculated correct

4Nightmares, 2022-04-23:
### Application Version

5.0.0 beta 1

### Platform

Mac OS Monterey

### Printer

Monoprice Cadet

### Reproduction steps

put one object on the plate and copy 3 times.

### Actual results

Raft is only calculated for the first copy.

### Expected results

Raft should be below all object (copies).

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Leinwand-Haken.zip](https://github.com/Ultimaker/Cura/files/8547571/Leinwand-Haken.zip)

File renamed from 3mf to zip for upload!

GregValiant, 2022-04-23:
Parts that have zebra stripes are outside the build area.  The dark border you see around the build area in Cura is the raft area.  Your parts are intruding into that area and so they can't be sliced as the raft would hang off the edge.
a)  You can set the location of all the parts to Y=0.  That will keep them out of the raft area.
b)  You can switch to the Custom settings and set the "Raft Extra Margin" to 10mm to make the dark border narrower.
Either way will allow the parts to be sliced.  In this image I've done both (narrow raft and all parts at Y=0).
[image]

4Nightmares, 2022-04-24:
Thank you very much for the quick help.
Just one more question: Shouldn't there be at least some warning for such cases? I remember having that sometimes on the 4.x versions.

GregValiant, 2022-04-24:
There is a popup for "Unable to Slice".  It still comes up when it's a single model that is placed so the skirt/brim/raft would print out-of-bounds.  You had an unusual problem in that 1 of the 4 models would slice.  It may relate to "raft" or something else.  At any rate maybe some particular set of circumstances negated the appearance of the popup - or the visual cue of the zebra stripes is considered enough of a warning?  I'm sure the Cura Team is busy running around putting fires out but someone will look at this.