Ultimaker/Cura #11913: UI font is too small compared to previous versions

tapir, 2022-04-23:
### Application Version

5.0 Beta

### Platform

Linux

### Printer

Voron

### Reproduction steps

Run the appimage version of the latest 5.0 beta

### Actual results

[image]

The fonts for the menu and the settings dialog is too small.

### Expected results

Font size should either respect system font settings or match whatever it was in previous 4.9 version.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

chipmunk1977, 2022-04-24:
What distribution and version of Linux are you using?  I'm using Ubuntu 20.04LTS and the font size and spacing increased.

tapir, 2022-04-24:
Arch linux with latest gnome.