Ultimaker/Cura #11915: Default save directory is unaccesible

tapir, 2022-04-23:
### Application Version

5.0 Beta

### Platform

Linux

### Printer

Voron

### Reproduction steps

1- Slice a file
2- Try saving it

### Actual results

[image]
As seen it tries to save to "/home" directory which the current user has never write access.

### Expected results

It should try to save to at least "/home/{USER_NAME}" or better the last directory it remembers where the save happened.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

fvrmr, 2022-05-04:
Hi @tapir thank you for your report.
Could you also share your log file? You can find your log file here: `$HOME/.local/share/cura/<Cura version>/cura.log`

tapir, 2022-05-04:
Sorry this was probably a Gnome issue and was fixed on one of my regular system updates. Cura now opens up the home directory by default. Only annoying thing is that even though it filters gcode extension by default, when you enter a filename it doesn't save it with the gcode extension.

closed 2022-05-04 by tapir