Ultimaker/Cura #11918: Cura-5.0.0-beta+1 failed start with firejail

sppmg, 2022-04-23:
### Application Version

5.0.0-beta+1

### Platform

Linux (Debian 11)

### Printer

NA

### Reproduction steps

Cura-5.0.0-beta+1 can't run with firejail (0.9.68), **but Cura-4.13.1 work fine.**

```
firejail --noprofile --appimage ./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage
```

### Actual results

It's resule of Cura-5.0.0-beta+1 with firejail.
```
Parent pid 924531, child pid 924534

** Warning: dropping all Linux capabilities and setting NO_NEW_PRIVS prctl **

Mounting appimage type 2
Warning: NVIDIA card detected, nogroups command ignored
Child process initialized in 58.51 ms
/bin/bash: line 1: /run/firejail/appimage/AppRun: Permission denied

Parent is shutting down, bye...
AppImage detached
```

### Expected results

Expected results just work.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Maybe appimage package has some problem ?

probonopd, 2022-05-13:
Do earlier Cura AppImage versions work on the same system with the same Firejail?

sppmg, 2022-05-14:
Yes, Cura-4.13.1 work fine. firejail version `0.9.68` @ Debian11 .

sppmg, 2022-05-14:
Append Cura 4.13.1 log.

```
$ firejail --noprofile --appimage ./Ultimaker_Cura-4.13.1.AppImage 
Parent pid 291109, child pid 291112

** Warning: dropping all Linux capabilities and setting NO_NEW_PRIVS prctl **

Mounting appimage type 2
Warning: NVIDIA card detected, nogroups command ignored
Child process initialized in 48.15 ms
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
Cyclic dependency detected between "file:///run/firejail/appimage/usr/bin/resources/qml/Actions.qml" and "file:///run/firejail/appimage/usr/bin/resources/qml/Actions.qml"
WARNING: Cannot find style "default" - fallback: "/run/firejail/appimage/usr/bin/qt/qml/QtQuick/Controls/Styles/Desktop"

(window started)
(Close window)

Parent is shutting down, bye...
AppImage detached
```

probonopd, 2022-05-14:
Does Cura-5.0.0-beta+1 run on your same machine without Firejail?

sppmg, 2022-05-14:
Yes, It work.
```
$ ./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage 
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltimaSbnjKC/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimaSbnjKC/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
```

probonopd, 2022-05-14:
The file `AppRun` inside `Ultimaker-Cura-5.0.0-beta+1-linux.AppImage` has the wrong permissions. It should be `chmod 0755 AppRun`.

To see this:

```
# Working version

./Ultimaker_Cura-4.13.1.AppImage --appimage-offset   
188392

unsquashfs -ll  -o 188392 ./Ultimaker_Cura-4.13.1.AppImage  | grep AppRun
-rwxr-xr-x root/wheel            10872 2022-01-28 12:04 squashfs-root/AppRun

# Non-working version

./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage --appimage-offset
189632

unsquashfs -ll  -o 189632 ./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage | grep AppRun
-rwxrw-r-- root/wheel              460 2022-04-21 14:22 squashfs-root/AppRun
```

probonopd, 2022-05-14:
@sppmg here is how you can make an AppImage with fixed permissions:

```
./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage --appimage-extract
wget -c https://github.com/$(wget -q https://github.com/probonopd/go-appimage/releases -O - | grep "mkappimage-.*-x86_64.AppImage" | head -n 1 | cut -d '"' -f 2)\nchmod +x mkappimage-*.AppImage\nVERSION=
VERSION="5.0.0-beta+1" ./mkappimage-709-x86_64.AppImage squashfs-root
sudo chown -R $USER squashfs-root
chmod 0777 squashfs-root/
chmod 0755 squashfs-root/AppRun
VERSION="5.0.0-beta+1" ./mkappimage-709-x86_64.AppImage squashfs-root
```

Please let us know if it resolves the issue.

sppmg, 2022-05-14:
It work fine now (with firejail). Thanks your help.

probonopd, 2022-05-14:
I suspect that something needs to be done in

https://github.com/Ultimaker/cura-build-environment/blob/master/cmake/create-appimage.cmake

Need `0755` on `AppRun`.
Is this correctly translated to `FILE_PERMISSIONS WORLD_READ WORLD_EXECUTE OWNER_WRITE`?

Please see my PR @jellespijker
* https://github.com/Ultimaker/cura-build-environment/pull/140

jellespijker, 2022-05-16:
Hi @probonopd I saw your PR, thank you for your contribution. It still needs some additional changes before I can merge it. Unfortunately I don't have the time for this myself atm. If you manage to make these changes today or beginning of tomorrow, we might be able to get them into the stable release of 5.0

@fvrmr fyi

probonopd, 2022-05-16:
Changed the PR according to your feedback. Thanks!

fvrmr, 2022-05-17:
Thanks @probonopd for the PR. Closing this ticket since it should be fixed in 5.0 now!

closed 2022-05-17 by fvrmr

sppmg, 2022-05-18:
Hi, thanks your help. But `5.0.0` still has same problem.
```
$ firejail --noprofile --appimage ./Ultimaker-Cura-5.0.0-linux.AppImage 
Parent pid 631150, child pid 631153

** Warning: dropping all Linux capabilities and setting NO_NEW_PRIVS prctl **

Mounting appimage type 2
Warning: NVIDIA card detected, nogroups command ignored
Child process initialized in 68.45 ms
/bin/bash: 列 1: /run/firejail/appimage/AppRun: 拒絕不符權限的操作

Parent is shutting down, bye...
AppImage detached
```
Permissions of AppRun:
```
$ ls -l squashfs-root/AppRun 
-rwxrw-r-- 1 sppmg sppmg 460  5月 18 13:53 squashfs-root/AppRun
```

For note, it also solved after repack process.

probonopd, 2022-05-18:
@sppmg please check:

./Ultimaker-Cura-5.0.0-linux.AppImage --appimage-offset
<SOMENUMBER>

unsquashfs -ll  -o <SOMENUMBER> ./Ultimaker-Cura-5.0.0-linux.AppImage | grep AppRun
```

If the output is not `-rwxrw-r-- root/wheel` then it is broken.

sppmg, 2022-05-18:
```
$ ./Ultimaker-Cura-5.0.0-linux.AppImage --appimage-offset
189632
$ unsquashfs -ll  -o 189632 ./Ultimaker-Cura-5.0.0-linux.AppImage | grep AppRun
-rwxrw-r-- root/root               460 2022-05-17 17:29 squashfs-root/AppRun
```