Ultimaker/Cura #11919: Long loading time for STL files | Binding loop detected

ComanderKai77, 2022-04-23:
### Application Version

5.0.0-beta+1

### Platform

Windows 10

### Printer

Custom FFF printer

### Reproduction steps

1. Start Cura
2. Download Benchy from [https://github.com/CreativeTools/3DBenchy.stl](https://github.com/CreativeTools/3DBenchy/blob/master/Single-part/3DBenchy.stl)
3. Drag Benchy into Cura

### Actual results

It takes over 30s to load Benchy. On Cura 4 it took less than 5s.

### Expected results

Benchy should load in less than 5s.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

`2022-04-23 22:00:53,930 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Program Files/Ultimaker Cura 5.0.0/share/cura/resources/qml/ObjectItemButton.qml:85:13: QML Button: Binding loop detected for property "width"` seems like the reason for Cura beeing slow. I didn't find the cause for the button binding loop.

[cura.log](https://github.com/Ultimaker/Cura/files/8548512/cura.log)

fvrmr, 2022-05-02:
Hi @ComanderKai77 thank you for your report.
Does this happen for every model? For me it loads in less than 5 seconds.

fieldOfView, 2022-05-02:
Binding loops that are detected are almost never the cause of low performance. When they are detected, Qt "cuts them off".

ComanderKai77, 2022-05-02:
Thank you for your response @fvrmr 
I've checked some bigger models (~10MB) like [https://cdn.thingiverse.com/assets/26/3c/36/1a/a4/yellow.stl](https://cdn.thingiverse.com/assets/26/3c/36/1a/a4/yellow.stl) and they also need around 30s to load.

I'm using a i5-7200U with 16GB RAM.
The cpu is half utilized (no core is fully utilized).

ComanderKai77, 2022-05-28:
In the current version of Cura (5.0.0) the issue seems to be fixed.
Benchy loads in under 5s.

closed 2022-05-28 by ComanderKai77