Ultimaker/Cura #11925: Filter Out Tiny Gaps causing Delta Printer to rattle like crazy

adamantiz, 2022-04-24:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Anycubic Predator Delta

### Reproduction steps

Slice any file which requires the Filter Out Tiny Gaps option
Also using Fill Gaps Between Walls = Everywhere

### Actual results

Using these options causes the Delta Printer to vibrate violently, to a point where screws rattle lose and it makes a lot of noise.
The only option then is to slow down the print to a point where printing takes forever. Needlessly printing the rest of the print at extremely slow speeds in order for the printing of tiny gaps - not to wreck the printer.

### Expected results

The slicer needs to slow down the speed of printing when filling tiny gaps, then speed it back up to regular speed afterwards.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The Anycubic Delta printer is capable of printing at speeds of upto 150mm/s depending on what hotend is being used. With a 0.4mm nozzle and standard(default) hotend, it can easily handle 90mm/s printing. However, when printing in the tiny gaps, 45mm/s still seems too fast.

Please will you edit this feature of the application so that the G-code produced will speed up and slow down the printer as necessary when using this option. Some prints demand this option for quality.

adamantiz, 2022-04-24:
should not require all prints to take 3 times longer, just to use this feature.

nallath, 2022-06-15:
Since we don't have delta printers, this isn't an issue for Ultimaker printers. Although we agree that this is an issue, it's not something that we will spend time on to fix.

closed 2022-06-15 by nallath