Ultimaker/Cura #11926: Cura 5 - Spiralize outer contour (vase mode) not working

MikeGlue, 2022-04-24:
### Application Version

5.0.0-beta1

### Platform

Windows 11

### Printer

Ender 3 v2

### Reproduction steps

1. Import attached "pineapple-vase" STL
2. Enable 'Spiralize Outer Contour' and slice

[pineapple-vase.zip](https://github.com/Ultimaker/Cura/files/8548970/pineapple-vase.zip)

### Actual results

The model is sliced with a lot of inner elements that should not be there.  

See pic below:

[image]

### Expected results

Only the outer wall should be printed, as shown below when the same STL is sliced in Cura 4.13.1:

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

..

smartavionics, 2022-04-24:
Try enabling "remove all holes".

fieldOfView, 2022-04-24:
... but don't forget to turn it off again when you are done.

fvrmr, 2022-04-25:
Hi @MikeGlue thank you for your report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/11843 and we have a ticket on our backlog for this. 
I will close this issue since it is a duplicate.

closed 2022-04-25 by fvrmr