Ultimaker/Cura #11928: Cura 5 beta Auto Closure

Babarakus, 2022-04-24:
### Application Version

5

### Platform

Windows 11

### Printer

Ender 3

### Reproduction steps

Clicked on page and just closed mid print via USB

### Actual results

above

### Expected results

expect it not to close app mid print

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Fix bug pls.

fieldOfView, 2022-04-24:
Provide logs pls.

We can't really know what happened without further information. And if we don't know what happened, we can't fix it so it won't happen again.

Babarakus, 2022-04-24:
Hey

Where aboits can I find the logs. I've uninstalled 5 and usint 4.13 again cos it's stabke and problem free. Will the 5 logs still be under the Cura install folder. 

Regards

MLocke13, 2022-04-24:
Hey, I am having the exact same issue.
I attached my .Log
[cura.log](https://github.com/Ultimaker/Cura/files/8550128/cura.log)

closed 2022-04-25 by fvrmr