Ultimaker/Cura #11929: [5.0 beta] Won't start anymore even though I used it succesfully fo a full day before

printingotb, 2022-04-24:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

I used it for an entire day, restarted succesfully several times. The next day it froze on startup; "not responding".

Cannot think of anything I did to cause it. No changes to preferences. No changes to computer. No computer restart.
Restarting computer does not help.

Reinstalling it did work. I also deleted the config folders just in case before re-installing it.

### Actual results

Won't start anymore

### Expected results

Should start

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8549249/cura.log)

fvrmr, 2022-04-28:
Hi @printingotb thanks for reporting.
There is nothing in the log why this is happening unfortunately. Maybe removing your preferences file could help.

fieldOfView, 2022-04-28:
Do you have multiple displays? If so, it may help to uncheck the "Restore window position on start" in the General preferences in Cura.

printingotb, 2022-04-29:
> Hi @printingotb thanks for reporting. There is nothing in the log why this is happening unfortunately. Maybe removing your preferences file could help.

Where do I find the preference file? Can't see anything named preferences in the config folders.

> Do you have multiple displays? If so, it may help to uncheck the "Restore window position on start" in the General preferences in Cura.

Thanks, I'll try that.

printingotb, 2022-04-29:
Renamed "Cura.cfg" and now it starts again
[cura.cfg.txt](https://github.com/Ultimaker/Cura/files/8588967/cura.cfg.txt)

fieldOfView, 2022-04-29:
That is what unchecking the "Restore window position on start" fixes. It makes sure that Cura starts on the primary display, and it works "better" after that on some systems.

printingotb, 2022-04-29:
I deactivated the right monitor and I could now access the Marketplace, which previously would also make Cura crash (believe forgot to report that). So there is clearly an issue with multiple monitors. It had no effect on issue #11864 however.
Another change is that with only one monitor active, Cura will self terminate a few seconds after it has frozen. With multiple monitors, I had to terminate it manually.

This also fixes issue  #11933
This also fixes issue #11869

printingotb, 2022-12-20:
What's the status on this? I find myself not using Cura 5 much because of this. I keep forgetting to deactivate my second monitor every time I need to and then it crashes and I loose work.
Is there any info I can provide that would help?
Monitors are two identical MSI Optix MAG241CP using the same settings on the same graphic card. Both HDMI connections. I have tried turning off Nvidia control panel and Gforce experience in case they did something with one of the monitors but it didn't help.
Having the displays in duplicate mode does not make Cura crash.
Win10

printingotb, 2023-06-10:
The issues I had with Cura freezing when opening menus seems to be gone in Cura 5.3.1 :smiley:
Both with "Restore window position on start" on and off.

printingotb, 2023-06-21:
Hmm, it started doing it again a few days later.

fieldOfView, 2023-06-21:
@printingotb can you try the 5.4 beta?
https://github.com/Ultimaker/Cura/releases/tag/5.4.0-beta.1

printingotb, 2023-06-21:
Sure.
Issue still here unfortunately. 

Test methodology:
Clicking manage printers menu to trigger issue.
Freshly installed 5.4 beta.1
Initially opens on Main display

Everything works on the main display but it will crash if I move Cura to the secondary display and try to open a menu. If only one display is active, everything works regardless of which display.

Tests give the same outcome regardless of the setting Restore window position on start except that Cura freezes on startup if it starts on the second display.

(Not sure about my previous observations, maybe I didn't keep track on which display I had Cura on. Same behavior in 5.3.1 now).

printingotb, 2023-06-21:
Also, with Cura in widowed mode and the menu not fixed to the main display, it seems that it crashes if the window is more than 50% in the secondary monitor when opening the menu.

nallath, 2023-06-21:
Sniff :( We were really hoping that the qt upgrade would fix this.

printingotb, 2023-06-22:
Hey, if you can create special versions of Cura with possible fixes and want me to test them, let me know!

nallath, 2023-06-22:
I know and thanks. But right now, we're back to the "why the fuck is this happening in the first place" part of the problem