Ultimaker/Cura #11930: [5.0 beta] Tooltip doesn't disappear and tooltip is not applicable to my machine profile.

printingotb, 2022-04-24:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

Not sure unfortunately. Can't make it happen again. I think I was just opening and collapsing categories one by one eyeing over my settings.
Visibility is enabled for all settings but this one is hidden by the machine configuration.

### Actual results

A tooltip about inner wall extruder appeared and stayed until I hover over a different tooltip. Clicking on it or anywhere else did not remove it.
My machine profile doesn't have more than one extruder so this tooltip shouldn't be possible to trigger?

### Expected results

Dissapear by itself.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2022-04-25:
Hi @printingotb thank you for your report.
Strange, I can't reproduce it myself. Will have to keep this issue open and see if other people run into this issue.

printingotb, 2022-04-29:
Happened a couple of times more. Maybe it happens if I click slice before the tooltip have had time to disappear or at the exact  moment it triggers? I'm still not able to reproduce it when I try.
[image]

printingotb, 2022-05-07:
I find a way to reproduce it.
It happens when I click the revert changes symbol and immediately move the cursor down to an area that doesn't have any tooltips. The setting that you click need to be at the bottom of the list to be able to move away fast enough.
[image]

I noticed one more thing:
If I hover the mouse cursor on the fx symbol, the "this setting is normally caclulated..." tip does not appear. 
It does however appear after clicking the restore arrow while leaving the cursor in place.
The "...Click to Restore value the value of the profile" tip does not appear at all in 5.0 beta.
[image]

printingotb, 2023-02-23:
Got this today in 5.2.1 where it displayed a tooltip for a setting that was not visible. Raft was not enabled and the Raft base Wall count was as a result not visible. So I couldn't have moved the cursor over it. If it had been visible, it would also have been two rows below the last setting visible on the screen. With Remove raft inside corners on the first row.
It's almost as if there is a short delay before the tooltip function notices when the cursor leaves the settings list? And during this delay, it assumes that all settings are visible and generates tooltips for them?
I'm using the settings guide plugin if that's what causes it.
[image]