Ultimaker/Cura #11933: [5.0 beta] Cannot open projects

printingotb, 2022-04-24:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

Drag and drop or import projects 

### Actual results

causes Cura 5 to freeze

### Expected results

shouldn't

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[test.3mf.zip](https://github.com/Ultimaker/Cura/files/8549312/test.3mf.zip)

fvrmr, 2022-04-26:
Hi @printingotb thank you for your report.
I can open your project file, could you also share your log file? Does this happen every time?

printingotb, 2022-04-26:
Yeah I tried a couple of times with different projects. Cura stops working every time. 
This is the current logfile:
[cura.log](https://github.com/Ultimaker/Cura/files/8564405/cura.log)

Is the "cura.log" file the only relevant file or do you also want the "cura.log.1" or "cura.cfg" files?

Right now, issue #11929  happened again so I have to backup my profiles and re-install before I can recreate the issue with a fresh log.

fvrmr, 2022-04-28:
Hmmm there is also nothing in the logs here. Do you have a network disk?

GinoAnth, 2022-06-14:
Same problem,
Opening a saved project, Cura 5 goes in 'Not responding'
same 3mf project opens fine in Cura 4.13.1

Also dublicating an item makes cura 5.0.0 not responding