Ultimaker/Cura #11934: [5.0 beta] Inside to outside doesn't always work

printingotb, 2022-04-24:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

Slice a benchy, or my project. Look at layer 196 for example

### Actual results

The chimney is printed outside in but not the roof.

### Expected results

Should print inside out.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[3DBenchy.3mf.zip](https://github.com/Ultimaker/Cura/files/8549327/3DBenchy.3mf.zip)

fvrmr, 2022-05-02:
Hi @printingotb thank you for your report.
I will bring it up with the team. Keep you posted.

pkuiper-ultimaker, 2022-05-11:
Hi printingotb,

I must admit that this is very confusing, but it is intended behavior. The green line is what we call a center last line. It is not an inner wall. It could have been a inner wall of any of the two outer wall. The center line is always printed last in Cura.
I will make a ticket to update the help text for Cura 5.1. Has this answered your question?

printingotb, 2022-05-12:
Yes, thanks for the explanation.

> The green line is what we call a center last line.

Is there a plan to add this to the line type colour scheme? I believe that would be helpful as the middle line is a new concept to users.
Or at the least a different shade of green, like how the travel lines are light blue/dark blue, with/without retraction.

pkuiper-ultimaker, 2022-05-12:
Hi printingotb,

I created a ticket on the backlog (CURA-9290) to change the Wall Ordering description text and consider changing the line color in the slicing preview. Thanks for reporting this.  
Kind regards,

Paul Kuiper

printingotb, 2022-07-02:
Would it be possible to have the center line printed first in the cases where it is surrounded only by outer walls like above?
Outer walls have a tendency to not connect properly if printed first. The start of the outer wall will often be dragged along with the nozzle for a tiny distance, causing a gap in the wall. This is most noticeable after a few layers with the seam in the same position and smaller line widths are more sensitive to this.
[image]

pkuiper-ultimaker, 2022-07-04:
We have discussed a separate setting for the center line strategy at some point, but we could not find a reason to add such an setting and decided to always use center line last. I do not directly see a reason to change this. This seems to be related to start/stop behavior around a seam that you could optimize differently. You will have to same issue again when printing a single line wall when we would add such a feature.

puffofsmoke, 2022-09-11:
> We have discussed a separate setting for the center line strategy at some point, but we could not find a reason to add such an setting and decided to always use center line last. I do not directly [...]

Imagine a model (some toolbox for example) contain only this type of walls with 1.5mm wall width and 0.6 nozzle 

Start to sound cool to print centerline first and outer perimeter with an "Outer wall inset" option

[image]

MariMakes, 2022-11-01:
A new, more detailed description has been added to this setting to avoid confusion.

To me, it doesn't seem like a bug in Ultimaker Cura anymore. 
Would you still like to keep this issue open or do you agree that we can close it?

printingotb, 2022-11-01:
Sounds good to me, thanks for asking!

MariMakes, 2022-11-01:
Awesome. Thanks 🎉

closed 2022-11-01 by MariMakes

nateran48, 2024-07-25:
Please keep trying to fix this. I understand that the center line is different from an inner wall, but it really seems like if you are printing a part the thickness of 3 extrusions, I should be able to re-order them to have the middle print first. I want this because there is a gap after every retraction, and I prefer that gap be on the inside of the part.