Ultimaker/Cura #11937: drag and drop not working

arcadeworkshop, 2022-04-24:
### Application Version

5.0

### Platform

Windows 10 Professional

### Printer

Creality Ender 3

### Reproduction steps

Opened Cura 5.0. Opened File Explorer. Tried to drag STL file from file explorer to Cure 5.0. Drag and drop did not work.

### Actual results

STL file does not transfer to Cura 5.0. Instead, the cursor turns to circle with diagonal line.

### Expected results

STL file should appear in Cura 5.0, but does not.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

tispokes, 2022-04-25:
Same here, wanted to submit a bug report now.
@arcadeworkshop maybe it would be better to add "v5b1" or something into the title easier for devs to see.
@cura-team Maybe add a label for cura 5 beta, easier to filter for you :-)

arcadeworkshop, 2022-04-25:
> Hi @arcadeworkshop <https://github.com/arcadeworkshop> thank you for your
> report.
> This is a known issue, I have encountered this myself as well. Do you use
> an external monitor? For me it doesn't happen when I have my monitor not
> connected.
> Devs see CURA-9156
>
> @tispokes <https://github.com/tispokes> thanks for the tip :)

I'm running on a desktop, so I do have a monitor. New laptop should be
arriving this week, so I'll test on there as well.

tispokes, 2022-04-25:
@fvrmr **Today** I can use drag and drop in v5b1 on both screens. Only i switched between primary and secondary for some screenshots to submit another bug. Then just used drag and drop without thinking about and it worked well. On both screens.

arcadeworkshop, 2022-04-25:
I just tried again and now drag and drop seems to be working. Not sure what changed, other than the system being restarted in between.

nallath, 2022-04-26:
So basicly; It seems to not work the first time you run Cura? Can people confirm that that is the behavior?

If so, that will make it a lot harder to fix it, since it's pretty hard to reproduce that way :(

fieldOfView, 2022-04-26:
Was the NSIS installer script overhauled when the build system was changed? Something must have changed, because the main executable name has changed.

If the installer starts a Cura instance, it may be started with elevated privileges ("as administrator"). You can not - as an unelevated user - drop files into an elevated process. We had this issue before, and the solution was to not run `Cura.exe`, but run `explorer.exe` with an argument to have explorer open Cura.

fieldOfView, 2022-04-26:
See https://github.com/Ultimaker/cura-build/pull/121 for the previous fix.

closed 2022-05-04 by nallath