Ultimaker/Cura #11941: (5.0) "Manage Printers" and "Machine Settings" dialog issues

GregValiant, 2022-04-24:
### Application Version

5.0.0 beta

### Platform

Windows 10 Home (onboard video OpenGL 4.0 in compatibility mode)

### Printer

Creality E3Pro

### Reproduction steps

The Enter (default?) and Cancel buttons are missing from both dialogs.  The "dismiss" X on the titlebar has to be used to close the dialog.  There is no indication if the settings have been accepted/saved and no way to discard any changes.

When selecting multiple lines of StartUp or EndGcode - the selected text is no longer highlighted to inform the user that they have indeed been selected.  The "Extruder" textboxes work as expected.

There is no right click menu to copy or paste in any text box in Machine Settings.  CTRL+C or CTRL+V work for copy and paste.  Previous versions suffered from this as well.

### Actual results

See above.

### Expected results

Normal dialog and text box interaction.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Log File
[cura.log](https://github.com/Ultimaker/Cura/files/8549932/cura.log)
.

GregValiant, 2022-04-25:
Additional:
When making changes to either StartUp gcode or Ending gcode, if the "close" X button is immediately clicked to dismiss the dialog then the users' changes are not made.
If the user clicks on any other area first (or tabs out of the text box) then the changes are made when the dialog is dismissed.

Vandresc, 2022-04-25:
Devs, see CURA-9188 for our internal reference.

GregValiant, 2022-04-26:
I am not able to export profiles.
Importing is OK.

fvrmr, 2022-05-03:
The ticket on our backlog is has been closed, since the close button and x should do the same. So that is why the close button is removed. However there is an issue with no saving the values. I created a ticket for that, which is linked to https://github.com/Ultimaker/Cura/issues/12021

About selecting text in the machine settings I will bring that up with the team. 

The right click menu is a duplicate of this: https://github.com/Ultimaker/Cura/issues/11579 and we have a ticket on our backlog for this.

GregValiant, 2022-05-03:
Thank you @fvrmr .

Madmike-1337, 2022-07-22:
There are about 30 tickets for this issue spread over several versions. It makes tuning, troubleshooting, and general setup of any Non-Ultimaker printer more difficult to get using. Why is it such a pain for the users and such a low priority for the devs?

nallath, 2022-07-22:
> There are about 30 tickets for this issue spread over several versions. It makes tuning, troubleshooting, and general setup of any Non-Ultimaker printer more difficult to get using. Why is it such a pain for the users and such a low priority for the devs?

Well, you already hinted to the reason in your own post. As you might be aware, the vast majority of Cura development is being paid for by Ultimaker. As such, fixes /changes that benefit Ultimaker users will automatically get more priority. That obviously doesn't mean that they won't be fixed, but that it could take longer.

GregValiant, 2022-07-28:
Just for an update - this behavior is still present in 5.1.0.

Vandresc, 2022-07-29:
We created a ticket in our backlog, there is still some weirdness there indeed. Thank you for the update @GregValiant.
Devs see CURA-9523, for our internal reference.

Madmike-1337, 2022-07-29:
> We created a ticket in our backlog, there is still some weirdness there indeed. Thank you for the update @GregValiant. Devs see [CURA-9523](https://ultimaker.atlassian.net/browse/CURA-9523), for our internal reference.

I have some observations that may be helpful to the dev team. Is this the right place to put them?

Vandresc, 2022-08-01:
If it is related to this ticket, sure! Thank you

Madmike-1337, 2022-08-01:
> If it is related to this ticket, sure! Thank you

The problematic behavior seems to be related to where Cura stores the user changes.
When going from a past version to a new version Cura seems to store and retrieve printer and material settings from previous versions. e.g. When I went from 5.0.0 to 5.1.0 all of my settings in AppData were still being changed in the 5.0.0 directory while leaving the settings in the 5.1.0 directories untouched. 

As part of troubleshooting, I backed up all of my printer and material settings then removed all past version folders from C:\Users\madmi\AppData\Roaming\cura and left the folder for 5.1 in place. 

After removing this folder then restarting, cura recommended a reset of user settings. I did so and then restored from my backed up files.

After the reset and restore, I can now make changes by clicking the X to close the editor window. 

I have not downloaded a past version and attempted replication.

InstantMuffin, 2023-04-20:
I'd like to open up a new issue if wanted, but I just wanted to mention that this does not seem to be fixed (for me).
Using latest Cura on Ubuntu as an AppImage, Cura just does whatever it likes with both the machine settings and the print profile settings (Yes, I know about the settings import dialog when opening up previously saved projects etc.).