Ultimaker/Cura #11943: Hot end and bed Heating simultaneously

Savage79, 2022-04-24:
### Is your feature request related to a problem?

Title says it best, but Cura is the only slicer I can see that doesn't offer this feature.  It would be nice to have the option to either heat independently or simultaneously. 

### Describe the solution you'd like

Option to heat all heaters at same time

### Describe alternatives you've considered

using other slicers.....

### Affected users and/or printers

all

Asterchades, 2022-04-25:
You can do this by adding heating commands to the Start GCode. If Cura detects them in said code, it won't add extras. This is also a good way to get it to not heat at all should you prefer to pre-heat instead by adding the lines and commenting them out.

`M104 S{material_print_temperature_layer_0} T{initial_extruder_nr}
M190 S{material_bed_temperature_layer_0}
M109 S{material_print_temperature_layer_0} T{initial_extruder_nr}`

This will start the first tool heating (or only tool if you have multiple) to the desired first layer temperature but continue parsing code. Then it will set the bed to the first layer temperature but not progress further until it has reached that temperature. The final line should be unnecessary but is for insurance, and will wait until the first tool reaches the first layer temperature before moving on (it shouldn't be necessary as the tool generally heats much faster than the bed, despite the higher temperature).

tispokes, 2022-04-25:
Same is possible with the bed. Check M140 to set a bed temp. M190 is waiting.
So it's not curas fault. more your start g-code.

I prefer setting the desired temp of bed and nozzle (M190/M140) and then doing other stuff as homing X/Y and then when its fully heated (M104/M109) to home Z, then draw the purge line and start printing.