Ultimaker/Cura #11947: 5.0.0-beta+1 Wall Printing Issue

fefifofums, 2022-04-24:
### Application Version

5.0.0-beta+1

### Platform

Windows 10

### Printer

Ender-3  Pro v2

### Reproduction steps

Using 5.0.0-beta+1 Ceality Ender-3 Pro all default settings  except 70mm print speed
1. Slice [object](https://cdn.thingiverse.com/assets/76/ec/16/21/8a/Adaptapod_Neck_v1.stl)
2. Print
[image]
3. See result

Note - Same result multiple times with different filament temperatures. I have not tried slowing down the print speed yet but I will update this section with the results.

Using 4.13.0 Ceality Ender-3 Pro all default settings  except 70mm print speed
1. Slice [object](https://cdn.thingiverse.com/assets/76/ec/16/21/8a/Adaptapod_Neck_v1.stl)
2. Print
[image]
3. See result

### Actual results

Results from printing on 5.0.0-beta+1 
[image]

### Expected results

Result of print on 4.12.0
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm happy to include any additional information such as the project file.  However, I will not that the previews look identical between both versions.   Please let me know if I can provide any other useful information.

GregValiant, 2022-04-25:
Post a project file and post that "bad" gcode file also.  I've got the same printer and we'll see if I get the same result.
Did you look at the preview with the Travel Moves turned on?  Settings like "Max comb distance with no retract", "Retraction Minimum Travel" and the "Combing" mode are the same in both Cura versions?

fefifofums, 2022-04-25:
I added the bad project file and gcode above.  While three settings you asked about match in both 5.0.0 and 4.13.0, it appears I did change the default values.  

**5.0.0 Preview** 
[image]

**4.13.0 Preview**
[image]

GregValiant, 2022-04-26:
Changing the combing mode to "All" seems to be the ticket.  Changing "Max comb distance with no retract" to 3.0 may help.  It's a small part.
Telling it to "Avoid Printed Parts" can have odd effects when there is only one part on the build plate.  It can make for travel moves outside the part and that can cause stringing.
I'm not a fan of "Retract before outer wall" as it can leave a blob when it primes at the start of the outer wall extrusion.  That can make the Z seam more noticeable.

fefifofums, 2022-04-27:
Interesting, I'll try change that and report back with my results.  

I'm still curious why I don't see this issue with the same settings on 4.13.0 though.  Any thought's on that?

fefifofums, 2022-04-27:
I figured out what the problem is.  In Cura 4.13.0, there is a setting in Walls called "Outer Before Inner Walls", in Cura 5.0.0-beta+1 this got changed to a setting called "Wall Ordering".  The problem is that in Cura 5.0.0-beta+1 the default value changed from "Inside to Outside" to "Outside to Inside" so when it was printing that first outer wall of the overhand, it had nothing to bond to, resulting in the drooping.  

### Solution - Set "Wall Ordining" to "Inside to Outside".

closed 2022-04-27 by fefifofums

GregValiant, 2022-04-27:
Well that makes sense.  I learned something as well.

fefifofums, 2022-04-27:
All good! It was helpful to have someone to talk to about it.  Thank you!