Ultimaker/Cura #11949: Cura 5.0 Beta Mac with Wacom

3dHawwy, 2022-04-24:
### Application Version

5.0 Beta

### Platform

macOS Monterey 12.3.1 Apple M1

### Printer

Reality Ender 3 V2

### Reproduction steps

When opening Cura 5.0 it is not responding to Wacom tablet input.  So, none of the menu's or anything else will respond when using the Wacom.  I can operate Cura 5.0 by using the Apple touchpad.  

### Actual results

No control over Cura 5.0 when using Wacom

### Expected results

Full control like all previous Cura versions

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No further comments required

Ghostkeeper, 2022-06-01:
Duplicate of https://github.com/Ultimaker/Cura/issues/2515.

MariMakes, 2022-08-23:
Whoops.
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12554 instead.