Ultimaker/Cura #11950: v5 beta1 crashes halfway thru print

carbon420, 2022-04-25:
### Application Version

5.0.0 beta 1

### Platform

windows 10

### Printer

Anycubic i3 Mega S/Pro

### Reproduction steps

sliced the file and started the print.  halfway thru the print, the printer just stops and the program has crash pop ups over and over and it freezes and crashes

### Actual results

app crashes 

### Expected results

the app shouldnt crash

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8550938/cura.log)

GregValiant, 2022-04-25:
Just to confirm...You are printing via the USB port?

fvrmr, 2022-04-25:
Hi @carbon420 thank you for your report.
If you are printing via USB this is a duplicate of https://github.com/Ultimaker/Cura/issues/11896
I will close this issue, but let us know if this happens without USB printing.

closed 2022-04-25 by fvrmr