Ultimaker/Cura #11952: Cura 5.0 beta 1 install forces delete of current configuration

Anathema-Device, 2022-04-25:
### Application Version

5.0.0.beta.1

### Platform

Windows 10

### Printer

n/a

### Reproduction steps

On a windows 10 machine with Cura 4.13.1 (and previous versions) installed and working with 3 printer and many material definitions the install of Cura 5.0.0.beta-1 forces the removal and reset of the configuration in order to run.  This means that the complete configuration of Cura 5 needs to be done again.  FWIW I saw this same behaviour with the Arachine beta as well and did not make much use of it due to this.

### Actual results

Cura creates a 5.0 config directory containing the printer and material definitions from Cura 4.13.1 but then fails to read them and forces the delete of the config before it can continue

Note: I have the .zip of the failing 5.0 config but this defect tool will not let me attach it

### Expected results

Cura 5.0 should create and use a config based on the existing 4.13.1 config without error

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8551234/cura.log)
[image]
[image]
[image]

nallath, 2022-04-25:
That specific crash has already been fixed for the 5.0 release. 

You can work around it by deleting your preferences file. The rest of your configuration should still be intact & working.

Anathema-Device, 2022-04-25:
I copied back the original 5.0 configuration folder content that was created at install and removed the cura.cfg file.  Then restarting Cura 5 resulted in a prompt to create a new printer config, not the config reset dialog.  I created (then deleted) a new printer and the result appears that the Cura 4.13.1 printers and materials have been brought forward.
Thanks!

closed 2022-04-25 by Anathema-Device