Ultimaker/Cura #11953: Open GL crash

ml878us, 2022-04-25:
### Application Version

5

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

Installed and tried to open

### Actual results

On my desktop (W10 64/i5/Lenevo) it wont open and gives an error about Open GL. On my laptop (W1065/i5/Dell) it opens fine. Both machines show drivers up-to-date.

### Expected results

Should open and run

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

groomit, 2022-06-10:
Same issue here on linux. Used to work on appImage 4.11.

OS:
Pop!_OS 22.04

Cura Version:
5.0

nallath, 2022-06-10:
Duplicate of #12359

closed 2022-06-10 by nallath