Ultimaker/Cura #11954: Printing Temperature Changes

dcs3473, 2022-04-25:
### Application Version

5.0 Beta

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

When using the Material Settings for 
Print Temperature
Print Temperature Initial Layer
Initial Printing Temperature
Final Printing Temperature

### Actual results

No matter what you change any of these to there is never a change in the Gcode.  They all use the Print Temperature Setting.

Same goes for Build Plate Temperature and Build Plate Temperature Initial Layer

### Expected results

There should be additional command in the Gcode to make changes to temperatures after the initial layer.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

GregValiant, 2022-04-25:
A project file may give clues as to what is going on with that.
Set Cura up to slice a model the way you want and then use the File | Save Project command to generate a project file.  Zip it up and post it here.

GregValiant, 2022-04-25:
What I am seeing here is that "Printing Temperature" and "Printing Temperature Initial Layer" work as do both build plate temperature settings.  @dcs3473 there may be something in your StartUp gcode that is having an effect.

Neither "Initial Printing Temperature" nor "Final Printing Temperature" appear to work.

dcs3473, 2022-04-25:
Here is my startup GCode for you to look at.  The only thing I do differently is preheat for bed leveling.

G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M301 P18.01 I1.57 D51.86 C0.01 ; Sock PID
M104 S160 ; set extruder temp
M140 S{material_bed_temperature_layer_0} ; set bed temp
M190 S{material_bed_temperature_layer_0} ; wait for bed temp
M109 S160 ; wait for extruder temp
G28 W ; home all without mesh bed level
G29 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
M104 S{material_print_temperature_layer_0} ; set extruder temp
M109 S{material_print_temperature_layer_0} ; wait for extruder temp
G92 E0.0 ; reset extruder distance position
G1 E-3 ; retract a bit
G1 X1 Y-3.5 Z0.3 F5000.0 ; move down to wipe ooze on bed edge
G1 X1 Y5 Z0.3 F5000.0 ; wipe
G1 X1 Y5 Z0.8 F3000.0 ; move to blob 1 position
G1 F200 E15 ;extrude a blob
G1 Z10 F200 E20; move the extruder up by 10mm while extruding
G1 X1 Y40 Z11.00 F10000; move away from blob
G1 X1 Y40 Z0.8 F3000.0 ; move to blob 2 position
G1 F200 E30 ;extrude a second blob
G1 Z10 F200 E35; move the extruder up by 10mm while extruding
G1 X1 Y60 Z.08 F10000; move away from blob
G92 E0 ; reset extruder

GregValiant, 2022-04-26:
That StartUp looks fine.
To utilize the "material_initial_print_temperature" you might have to call it out explicitly (instead of material_print_temperature_layer_0).  I don't really know.  Typically there will be an M104 line at some point before the completion of the first layer.
I think the "material_final_print_temperature" should be kicking in.  Maybe another setting is blocking it??  At any rate I haven't been able to get it to show up in the gcode.

fvrmr, 2022-04-26:
Hi @dcs3473 thank you for your report.
Could you share a project file (file - save project) and share it here as a .zip?

dcs3473, 2022-05-03:
I have not been able to duplicate this issue any more after the first two prints on Cura 5.0 Beta.  I am going to mark it down as a user issue.

closed 2022-05-03 by dcs3473