Ultimaker/Cura #11955: gfhfgh

freakydude, 2022-04-25:
### Application Version

5.0.0-beta1

### Platform

Windows 11

### Printer

Artillery Sidewinder X2

### Reproduction steps

1. Add/duplicate a custom filament. 
2. Change _Retraction_ to a value that is not whole number like (1.00, 2.00mm). 
3. For example, set it to 1.40mm

### Actual results

1. The Retraction is set to 1.00mm
2. After exporting the filament setting, the retraction is 1.00mm

### Expected results

1. The _Retraction_ is set to 1.40mm
2. After exporting the filament setting, the retraction is 1.40mm

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

**Further Test:**
Export Filament -> Edit Retraction to 1.40mm -> Import Filament again. Don't change Retraction ->  Export it, it's still 1.4mm. 
But after Edition Retraction, its a whole number again (1.0mm)

nallath, 2022-04-25:
Duplicate of #11840

closed 2022-04-25 by nallath

freakydude, 2022-04-25:
Sorry for duplicate and thx for merging

nallath, 2022-04-25:
> Sorry for duplicate and thx for merging

No problem :) I'd rather have things reported once too often than not at all. Thanks for testing it (and taking the time to report issues!)