Ultimaker/Cura #11957: UI problems. text seems to scale weird and is sometimes bearly readable.

PMvE, 2022-04-25:
### Application Version

5.0

### Platform

windows 10, 27 inch 4k monitor, gtx970

### Printer

Voron trident 250

### Reproduction steps

just installed the software stock install but the font and ui is kinda messed up i was using cura 4.12.1 before and the ui is fine. the problems are a lot of exsessive space in the print settings (this is not really a bug but more ineficient use of space). the font in some of the menu's seems to not scale correctly (this seems like a bug). my main monitor is 3840x2160 and the text is bearly readable. if i drag the window to my secondary screen 1050x1680 it is fine i made a screenshot so you can have a look. my 4K monitor is on the windows recomended 150% scaling and my secondary monitor is on the windows recomended 100% scaling.

### Actual results

text is not displayed correctly on a 4K screen but does on a 720p screen

### Expected results

text should be nice and readable on both resolutions directly after instaling.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
the left one is my 4K screen the right one my secondary low resolution monitor. the problem is properly visable at the words "HIPS" "CPE" "generic"

tispokes, 2022-04-25:
Same here on my 4K monitor

Pipsn4K, 2022-04-28:
I got the same problem on my 150% scaled 4K Monitor. With Cura 4.12. was all fine, with Cura 5 the UI is a mess. See screenshots:
Cura 4.12:
[image]
Cura 5:
[image]

fvrmr, 2022-04-28:
Hi @PMvE thank you for your report.
I will bring this up with the team, keep you posted.

tensiondriven, 2022-04-30:
My experience is probably related - I loaded Cura 4 and Cura 5 side-by-side on MacOS, external monitor at 1080p.  Taking a screenshot, I see this:

[image]

Left is Cura 5 Beta, right is Cura 4.

Issue: Cura 5 Beta text and glyphs appear soft, as if being scaled or aliased incorrectly.

mathieunicolas, 2022-07-17:
Hi, any news related to this bug ?

Boostedk10, 2022-12-18:
Any Updates on this I am also having this issue. makes the software look terrible

ColinLyn, 2022-12-24:
I have the same problem but I'm using two 1080p displays. 100% scaling.