Ultimaker/Cura #11958: (5.0) Print crashes

BarryKJ, 2022-04-25:
### Application Version

Cura 5.0 Beta

### Platform

Windows 10 x 64

### Printer

Original Prusa i3 MK3

### Reproduction steps

Installed Cura 5.0, selected a file and sliced as usual, printer stopped 5 minutes after start, no message displayed in pop up text box
[Cura 5.0 crash error.txt](https://github.com/Ultimaker/Cura/files/8554596/Cura.5.0.crash.error.txt)

### Actual results

Cura closed after error was displayed
[Cura 5.0 crash error.txt](https://github.com/Ultimaker/Cura/files/8554658/Cura.5.0.crash.error.txt)

### Expected results

Traced error log file and attached 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[Cura 5.0 crash error.txt](https://github.com/Ultimaker/Cura/files/8554617/Cura.5.0.crash.error.txt)

fieldOfView, 2022-04-25:
Duplicate of #11950, #11944, #11950, #11942, #11928, and last but not least #11896

MariskaMaas, 2022-04-25:
Oh, this is serious!
Thanks for reporting and for bundling the duplicates. 
For internal reference see CURA-9196

Megiddo62, 2022-04-27:
I am using a Tevo Tarantula and just downloaded v5 and it worked fine for my first print and then on the second one crashed after about 5 min.  The whole program closed itself so I could not get any screenshots.  Also not sure where to find the log files.