Ultimaker/Cura #11961: v5b1 Settings are inaccessible when switching from secondary to primary monitor

tispokes, 2022-04-25:
### Application Version

5.0.0

### Platform

W11

### Printer

-

### Reproduction steps

Cura 5 on secondary monitor (4k res) and the swithcing via Win+Shift+(Left | Right) to the primary monitor (FHD), the settings get cut off. Same appears when disconnecting the secondary while laptop is in sleep mode (lid closed).

E.g. I'm at home with primary and secondary monitor. Open Cura. Closing the laptop lid, disconnecting the secondary. When I then open the lid (e.g. moved to another location, train, whatsoever) then the settings are unrecoverable cut of and the print menu is not visible.

Even worse, while writing this - second bug. When I make the settings smaller on the secondary monitor, switch to the primary, then want to adjust the ssize of the settings correctly, the settings fold together and are not resizeable on the primary. Only back at the secondary.

### Actual results

4k: 
[image]

Switched to primary (FDH) with shortcut:
[image]

Or if i manually resize the settings beforehand:
[image]

When I now click and drag to resize on the small gray ribbon at the lower end with the 3 dots, the settings minimze it selfs like this (on the primary) and I'm not able to resize it on the primary:
[image]

### Expected results

Something like this.

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Also the more whitespace between every settings is disturbing me very well. I just have to scroll even more. More settings, more whitespace, more scrolling :-(

fvrmr, 2022-05-02:
Hi @tispokes thank you for your report.
I can reproduce your bug where the settings are cut off. About the minimizing issue, I think this is a known bug. I will bring both issues up with the team. Keep you posted. 

For the whitespace we have a ticket on our backlog!

fvrmr, 2022-05-03:
I have discussed it with the team and I created a ticket on our backlog for this. Devs see CURA-9241 For now it doesn't have a high priority, so don't except that this will be fixed in 5.0 stable.

As a work around you can resize Cura and then the print settings will reset and be accessible.

tispokes, 2022-05-03:
Ok. I even have it when I close cura (was on 4k), disconnect from hdmi, open again. Had it just right now.

Just mentioned for the backlog. Thanks.