Ultimaker/Cura #11962: Error beim slicen

RainR56, 2022-04-25:
### Application Version

5.0.0 beta 1

### Platform

Windows 10

### Printer

Geeetech A20M

### Reproduction steps

Ich wollte eine Münze drucken und bekomme beim slicen immer diese Fehlermeldung

### Actual results

[image]

### Expected results

[image]
[GA_Piece_Lucifer_V3.zip](https://github.com/Ultimaker/Cura/files/8555766/GA_Piece_Lucifer_V3.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[GA_Piece_Lucifer_V3.zip](https://github.com/Ultimaker/Cura/files/8555788/GA_Piece_Lucifer_V3.zip)

fieldOfView, 2022-04-25:
Please attach a logfile. We cannot diagnose issues like this without logs.
https://github.com/Ultimaker/Cura#logging-issues

RainR56, 2022-04-25:
Hello,here are the wanted log. RegardsRainer Gesendet von Mail für Windows Von: Aldo HoebenGesendet: Montag, 25. April 2022 19:33An: Ultimaker/CuraCc: RainR56; AuthorBetreff: Re: [Ultimaker/Cura] Error beim slicen (Issue #11962) Please attach a logfile. We cannot diagnose issues like this without logs.https://github.com/Ultimaker/Cura#logging-issues

GregValiant, 2022-04-26:
@fieldOfView - With the project file it appears that turning off Adaptive Layers allows it to slice.
The attached log has a failed slice, then Adaptive Layers turned off, then a good slice.
[cura log.zip](https://github.com/Ultimaker/Cura/files/8558852/cura.log.zip)

fieldOfView, 2022-04-26:
The relevant bit in the log is `[MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [954]: Backend exited abnormally with return code 3221225477`. That return code indicates that CuraEngine actually crashed because it did something "illegal" (like trying to access memory that it does not own).

GregValiant, 2022-04-26:
I forgot to add this last night...
Switching to my printer there was no problem using adaptive layers.
With the OP's printer - switching to a profile other than his "draft" profile and using adaptive layers, there was no problem.

fvrmr, 2022-04-26:
Hi @RainR56 thank you for your report.
I can also reproduce your issue, if I disable Adaptive layers like @GregValiant mentioned it will indeed slice correctly.

I will bring it up with the team. Keep you posted!

MariskaMaas, 2022-04-29:
We discussed it with the team and we'll have someone try to find a solution. 
For internal reference this is CURA-9228

fvrmr, 2022-05-04:
It seems like the issue is related to an other issue we had on our backlog. Devs see CURA-9178