Ultimaker/Cura #11964: Add variables for bed area used.... To drive new adaptive bed mesh capability in klipper

PhilBaz, 2022-04-25:
### Is your feature request related to a problem?

Yes, in a sense. Useful variable data is absent from the current code.

### Describe the solution you'd like

To have the required data to drive adaptive bed mesh native in Cura.

The Issue: 

From:  https://github.com/Frix-x/klipper-voron-V2/blob/main/macros/probing/bed_mesh.cfg

b. Cura is a bit more tricky as you need to install the post process plugin by frankbags called MeshPrintSize.py.
#          In Cura menu, click Help > Show configuration folder. Copy the python script from the following link into the plugins folder: https://gist.github.com/frankbags/c85d37d9faff7bce67b6d18ec4e716ff#file-meshprintsize-py
#               Then restart Cura and select in the menu: Extensions > Post processing > select Mesh Print Size
#               At the end, change your custom g_code PRINT_START macro to add the SIZE argument like this:
#               PRINT_START [all your shit..] SIZE=%MINX%_%MINY%_%MAXX%_%MAXY%

The current solution - python script.

From:  https://gist.github.com/frankbags/c85d37d9faff7bce67b6d18ec4e716ff

import re #To perform the search and replace.

from ..Script import Script

class MeshPrintSize(Script):

    def getSettingDataString(self):
        return """{
            "name": "Mesh Print Size",
            "key": "MeshPrintSize",
            "metadata": {},
            "version": 2,
            "settings":{}
        }"""

    def execute(self, data):
            minMaxXY = {'MINX':0,'MINY':0,'MAXX':0,'MAXY':0}
            lineData = ''

            for layer_number, layer in enumerate(data):
                for k,v in minMaxXY.items():
                    result = re.search(str(k)+":(\d*\.?\d*)",layer)
                    if result is not None:
                        minMaxXY[k] = result.group(1)

                areaStartGcode = re.search(".*%(MINX|MAXX|MINY|MAXY)%.*",layer)

                if areaStartGcode is not None:
                    if not lineData:
                        lineData = layer
                    for k, v in minMaxXY.items():    
                        pattern3 = re.compile('%' + k + '%')
                        lineData = re.sub(pattern3, v, lineData)
                    
                    data[layer_number] = lineData
        
                                         
            return data

### Describe alternatives you've considered

the current python script.

### Affected users and/or printers

Adapitve bed mesh should become the new norm. So more and more people.

parkernilson, 2024-02-24:
Has there been any progress on this? Is there a variable for first layer area in Cura?

PhilBaz, 2024-02-24:
@parkernilson My advice to you is to try Orca slicer.