Ultimaker/Cura #11966: [5.0 bea] Slicer fail with spiralize

redstone99, 2022-04-25:
### Application Version

5.0.0-beta+1

### Platform

Ubuntu 21.10

### Printer

XYZprinting da Vinci 1.0

### Reproduction steps

1. Open STL attached below
2. select "Spiralize Outer Contour"
3. line width 0.48
4. slice

### Actual results

Screenshot:  

[image]

### Expected results

I was expecting a smooth, continuous travel in upward spiral around the contour.  Like this:

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Zip with project file and STL:
[flor.zip](https://github.com/Ultimaker/Cura/files/8557479/flor.zip)

smartavionics, 2022-04-25:
Same as #11926 ?

GregValiant, 2022-04-25:
A hole instead of a pocket so pretty close.  The answer is the same.
For Spiralize to work correctly the model needs to be a solid.  Your model has that big ole' hole down the center.  The walls of that hole are also outside walls as they face air.
Under Mesh Tools select "Remove All Holes".  (Remember to turn it off when your done with it.)
This image is with Remove all Holes, Spiralize and 0 bottom layers.

[image]

redstone99, 2022-04-26:
I imagine that the slicer results I saw are not desirable in most if not all circumstances.  Seems like it'd be better if the slicer either threw up a warning or error dialog in this case with suggested fix of removing holes?

GregValiant, 2022-04-26:
An error reporting system can be useful when something breaks.  The very cryptic "Unable to Slice" would be infinitely more helpful if there was some explanation of Why the slice was unsuccessful.

_"...are not desirable in most if not all circumstances"_ In a nutshell - the "most but not all" is the problem.  How can a piece of software determine User Intent?  Should every setting change you make bring up a popup "Are you sure you want to do this?".  That would suck.

If you had the slice set up the way you wanted then the slice was correct and you wouldn't need a warning.  If after looking at the preview you say something like "Well that ain't right" then the settings need to be changed to make it right for your "Intent".  There isn't really an error to report, it's just that the slice is not what the User wanted/expected.
And today you learned something new about Cura.  That has to be worth something.

redstone99, 2022-04-26:
I appreciate wanting to avoid warnings, especially false-positive ones.  But it'd be nice to have the system better help (novice) users out.
Having the spiralize option dramatically change model geometry as in the screenshot seems unintuitive - I was expecting the feature to alter the tool path, not change model geometry.
The tooltip for the "make overhang printable" says that it can change the geometry of the model.
Maybe the tooltip for spiralize could mention that if there are holes in the part, spiralize might dramatically change model geometry?

I do appreciate learning about Cura!  I'm liking the 5.0 beta. Especially the smarter tool paths that the slicer generates (spiralize not withstanding)

fvrmr, 2022-04-26:
Hi @redstone99 thank you for reporting
Seems indeed related to https://github.com/Ultimaker/Cura/issues/11926, so I will close this ticket.
We have a ticket on our backlog for this. Will add your model as a test case!

closed 2022-04-26 by fvrmr